react-native-camera/react-native-camera #222: [enhancement] Ability to adjust quality and/or resolution

tkmoney, 2016-03-07:
It would be great if there was some way to adjust the image quality and/or the resolution of the image.

gm758, 2016-03-23:
Looks like this was added in the latest version. Super helpful for one of my current projects! Sending full resolution video over a network was a nightmare.

tkmoney, 2016-03-23:
Cool! It looks like there is a enum of strings options that you can pass to "captureQuality" but I dont see any documentation for what they are.

fgrs, 2016-08-23:
in my opinion this is a must have feature especially for those (including me) who want to store base64 images in AsyncStorage, currently i'm not able to store more then one image due to the AsyncStorage limit

charpeni, 2016-11-18:
I'll close this issue since it was introduced by #469, if you think it should still be opened let us know.

closed 2016-11-18 by charpeni