react-native-camera/react-native-camera #224: Requesting Android Permissions at Run Time?

rpastorelle, 2016-03-11:
Regarding [Requesting Permissions at Run Time](http://developer.android.com/intl/es/training/permissions/requesting.html)
> Beginning in Android 6.0 (API level 23), users grant permissions to apps while the app is running, not when they install the app.

Since this is using the "dangerous" permission: `android.permission.CAMERA`, shouldn't there be a corresponding permission check/request (say, `checkDeviceAuthorizationStatus`) in the Java module?

agrass, 2016-03-18:
Hi @rpastorelle, 

The open source version of React Native doesn't yet support the Android M permission. Is a known issue, but hopefully it will be solved by someone soon.  [https://facebook.github.io/react-native/docs/known-issues.html#android-m-permissions](https://facebook.github.io/react-native/docs/known-issues.html#android-m-permissions )

lucasferreira, 2016-04-08:
Hi folks,

I've made some experimental module to deal with this kind of permissions request of Android M:
https://github.com/lucasferreira/react-native-android-permissions

But you need to setup your * targetSdkVersion* to **23** to use them.

Best Regards ;)

rt2zz, 2016-04-08:
@lucasferreira doed sdkVersion=23 prohibit installation on devices < 23?

lucasferreira, 2016-04-08:
Nope @rt2zz

From the minimum target SDK 16 all of other Android versions can run your apps, even if they are built in 23 SDK version. 

By the way my module is very experimental and don't know if is possible to implement than inside other module like react-native-camera.

rt2zz, 2016-04-09:
Sure but at least we have a reference implementation that we can think about and experiment with. Awesome work!

randycoulman, 2016-10-03:
I note that react-native 0.33.0 has added the `PermissionsAndroid` API now.  That might make it easier to implement this. http://facebook.github.io/react-native/docs/permissionsandroid.html

punksta, 2017-05-31:
temp solution for that:

before opening of screen with camera, you can write something like that:
```
async goToQr() {
      // android 6.0 + 
      if ( Platform.OS === 'android' && Platform.Version >= 23) {
        try {
          const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA, {
            'title': 'We want use your camara',
            'message': 'reason of access '
          });
          if (granted === PermissionsAndroid.RESULTS.GRANTED) {
           goToCameraScreen(); // make camera open
          } else {
            Alert.alert(
               "can't access to camera title",
               "can't access to camera message",
                    [
                    {text: "ok" }
                    ]
            );
          }
        } catch (err) {
          //catch error here
        }
      } else {
       goToCameraScreen(); // make camera open
      }
    };
```

pvanliefland, 2017-06-07:
FYI, I got around this by creating my "own" camera module, consisting in a custom class that extends the Camera class provided by this module. The subclass thing is clearly not ideal, but if the library author is interested I can work on a pull request.

Here is a gist with the code I use: https://gist.github.com/pvanliefland/8c2522a27daf4759b015e3345ba47dda

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco