react-native-camera/react-native-camera #226: Very bad quality picture taken on android

KalebPortillo, 2016-03-16:
### Environment
- **React Native version**: 0.21
- **React Native platform + platform version**: Android 5.1

### react-native-camera
**Version**:  "master"

My code:
```
render() {
        return (
            <View style={styles.container}>
                <Camera ref={(cam) => {this.camera = cam;}}
                        captureTarget={Camera.constants.CaptureTarget.disk}
                        style={styles.preview}
                        orientation="auto"
                        aspect='fill'>

                </Camera>
                <View style={styles.hud}>
                    <View style={styles.hudHeader}>
                        <Image style={styles.hudHeader} source={this.state.photo}/>
                    </View>
                    <View style={styles.hudFooter}>
                        <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>
                    </View>
                </View>
            </View>
        )
    }
```
```
takePicture() {
        this.camera.capture()
            .then((data) => {
                this.setState({photo: {uri: data}});
            })
            .catch(err => console.error(err));
    }
```

So the problem is, when I run this in one of my android phones 4.4 kitkat, the image quality seens fine:
[image]

but when I run this in my Android phone 5.1 the image look like this:
[image]

Im running the exactly same code in both phones..
I dont know if i should retirve the image and do something with it before show in the image tag.
Thanks in advance

agrass, 2016-03-18:
I have the same issue in android

agrass, 2016-03-18:
I think it should be solved with this PR #211

KalebPortillo, 2016-03-18:
Thanks for help @agrass ,
Any idea when this PR will be accepted?

agrass, 2016-03-19:
The PR is in standby because they want to make the resolution more configurable. There are a solution proposed, but I have no knowledge of the progress of that, hopefully soon

closed 2016-03-23 by KalebPortillo