react-native-camera/react-native-camera #227: Saved image is incorrectly rotated - picture orientation does not match the preview orientation.

cdimascio, 2016-03-16:
### Steps to reproduce
1. Create react native app using sample code provide in README.md
2. Run app on a Nexus 9 (or any device whose default orientation is landscape not portrait)
3. Hold device in portrait mode and Take a picture. (Picture is now saved)
4. Open picture. The picture is rotated incorrectly. It's sideways

### Expected behaviour
Saved Picture should retain the same orientation it had when taken.

### Actual behaviour
Saved Pictured orientation is rotated incorrectly

### Environment
- **Node.js version**:  5.2.0
- **React Native version**: 0.21
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc Android 6

### react-native-camera
**Version**: npm version or "master"

rpopovici, 2016-03-22:
This is happening because on most SAMSUNGs and some other devices the orientation information is saved in EXIF metadata tag. You have to use this information if you want to properly visualize this picture. Take a look here for more info http://stackoverflow.com/questions/28298956/captured-image-getting-rotated-on-nexus-9

rpopovici, 2016-03-23:
Guys I made a PR which is suppose to fix the locked orientation problem. This might fix the Nexus 9 issue too https://github.com/lwansbrough/react-native-camera/pull/232

ljones87, 2017-08-23:
+1 any further resolution on this? Trying to work with react-native-camera on google pixel 7.1.2. Also utilizing react-native-orientation-lock in landscape mode..

ljones87, 2017-08-31:
Short term fix: we used the react-native-camera, orientation={Camera.constants.Orientation.landscapeLeft}, in conjunction with react-native-image-resizer: ex. ImageResizer.createResizedImage(data.path, 640, 480, 'JPEG', 100, 0)), the last number, 0, specifying the rotation, as well as react-native-orientation: Orientation.lockToLandscape().

mylestan, 2018-01-08:
Ran into this issue was well. The use of:

> react-native-image-resizer: ex. ImageResizer.createResizedImage(data.path, 640, 480, 'JPEG', 100, 0))

As suggested by @ljones87 worked for me.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

monolithed, 2018-02-05:
https://github.com/react-native-community/react-native-camera/pull/735