react-native-camera/react-native-camera #234: does this component support recording video on Android?

faceyspacey, 2016-03-26:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

vespakoen, 2016-04-15:
Not yet,

However, my initial stab at it can be found here:

https://github.com/vespakoen/react-native-camera/commit/3d0ee9475166d83d9bdb73dfb4cf3ba681e3dd48

I feel I have tried anything, but cannot get a playable video file (there is data in there for sure, it just doesn't play, not on android, not in VLC)

This might be due to my device(s), the Samsung Galaxy S4 Mini, and a Nexus 7.

I commented a lot of the code to hopefully get other people going on trying it out / play with it / report back with findings / results.

To try out my branch, run:

```shell
npm install --save git+https://github.com/vespakoen/react-native-camera.git#master
```

Helpful links:

http://developer.android.com/guide/topics/media/camera.html#capture-video
http://developer.android.com/reference/android/media/MediaRecorder.html
http://developer.android.com/reference/android/media/CamcorderProfile.html
https://android.googlesource.com/platform/development/+/4b737b6/samples/browseable/MediaRecorder/src/com.example.android.mediarecorder/MainActivity.java

If you want to help / test and report your findings, please do!

marcejohnson, 2016-04-16:
Some more testing options... here is my pull request for code I've been working on for awhile to support recording video on Android:
https://github.com/lwansbrough/react-native-camera/pull/262

- Supports same parameters and values as IOS, including captureMode, captureTarget, captureQuality, etc.
- Updated Readme.md
- Example code

Goal is "drop in" for compatibility with IOS... to make Android and IOS equivalent. I will continue to enhance this as bugs or additional changes are needed since I need this for my own project.  Please send me you feedback, thanks.

marcejohnson, 2016-08-29:
The feature is officially merged and now part of the repo:
#262

You can close this.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco