react-native-camera/react-native-camera #236: Preview Screen Disappears after one second

sscaff1, 2016-03-30:
When you take a picture, a preview of the image comes up for one second and then begins capturing again (in still mode) so that you can take another picture. This is fine for certain use cases, but it would be nice to give the Camera Component the option of staying on that preview so that you can introduce a UI above the previewed image (similar to how SnapChat does it). I don't currently see this as an option for this package.

closed 2016-03-30 by sscaff1

lijax, 2016-05-18:
Same issue here，how did you solve it?

sscaff1, 2016-05-18:
@lijax There was no way with the package itself that I could find. All I did was on completion of the promise, I navigated to a scene with the image I just took - kinda hacky but it works.

lijax, 2016-05-19:
Thanks @sscaff1 . I almost did the same. Just rerender an Image to replace Camera element base on an image Path state.