react-native-camera/react-native-camera #237: Long Delay between onPress and Finished Callback

sscaff1, 2016-03-30:
There is an average of a 2 second delay on Android between pressing "Capture" and the data being returned from the promise. I used another package https://github.com/NorthFoxz/react-native-camera-android and there is an average of a 0.8 second delay for the same scenario (which is much better). Any ideas? I saw a closed issue from a while ago showing that the delay was related to a similar delay and it was associated with rotation of the image. So the current pull request (#232) by @rpopovici may fix the issue...?

cyphire, 2016-07-18:
I'm having this issue to... My app is dependent on being able to take picture after picture quickly.  My time between capture and onPress is more like 4 seconds.  I had to change my app to select "capture", create an array of the current pictures in the default camera location,  have user go to the regular camera, shoot a bunch of pictures, then go back to my app and re-press the "capture" button, selecting all images NOT in the array for processing.  This works but is cumbersome.  My normal response in the native camera are .4-.75 seconds per press.

npomfret, 2017-05-23:
Are these all android issues?  If so, I suspect I know what's causing it.  

To test the theory, could you try my [PR](https://github.com/lwansbrough/react-native-camera/pull/716) and set the target to anything other than cameraRoll (like `Camera.constants.CaptureTarget.temp`).

Then, if the delay is noticeably less, switch to CameraRoll and see if the delay comes back.

mnlbox, 2017-06-28:
I have same issue with Nexus 7 Tab (Android 5.1.1)

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco