react-native-camera/react-native-camera #241: Issues including images outside of the component where they were captured

evanmrose, 2016-04-01:
I have an application component which takes and orders a number of pictures and saves them to disk. I can reference them from within the same component and display them to the user. Outside of that view, however, I get errors saying module unrecognized. 

Is the path that is returned from the RN Camera callable from anywhere component in an application? It doesn't seem to be, but is it relative?

Has anyone else had an issue using images saved to disk from the camera in Image tags?

React Native .21
iOS 9.2

evanmrose, 2016-04-06:
Hey @lwansbrough also setting captureTarget to the camera roll doesn't return a base 64 encoded image as per the docs, it returns `assets-library://asset/asset.JPG?id=100D4B73-1ACF-4A83-9960-7CAB2329383B&ext=JPG`

closed 2016-04-21 by evanmrose

albenik, 2016-05-18:
@evanmrose How do you resolve this issue?