react-native-camera/react-native-camera #242: when trying to run in device getting error

shaikhussian, 2016-04-02:
hai when i try to run this in device, I am getting an error like the below screenShot,when click on capture
[image]
can you please tell me that how to solve it, And i am using this module for QR-Reader, how can i scan the generated QR  using this module

rt2zz, 2016-04-02:
Most likely you are using an old version that haf a callback interface for capture. I would try installing master and see if that helps

shaikhussian, 2016-04-04:
I am using these versions
react-native: 0.22.2,
react-native-camera: 0.3.8,

shaikhussian, 2016-04-04:
@rt2zz just i saw saw your master and changed the node_module  according to that it show error like this in Camera.js

[image]

rt2zz, 2016-04-04:
you will need to rebuild your project since the objC source changed. Also should clear your caches just in case

shaikhussian, 2016-04-05:
@rt2zz  i rebuild my project again,camera is working but when i move camera to generated qr code it shows some error message like this in device
[image]

rt2zz, 2016-04-05:
After cursory inspection, i think there may be a bug in that onBarCodeRead is called even if it us not defined. 

1) try specifying a onBarCodeRead prop function
2) if this is the problem a PR to default onBarCodeRead prop would be much appreciated

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco