react-native-camera/react-native-camera #246: 1.0 Changes, TODO, and RFC

rt2zz, 2016-04-04:
So far the following changes are slated for 1.0
- Promise api instead of callbacks
- Android Support
- `mirrorImage` prop
- `captureQuality` prop

Discussion
- Option to set jpeg quality (possibly android only?)
- Return value for capture to become an object with metadata instead of a simple string value

TODO
- Android Video
- Android Barcode
- Android support child views
- iOS - ImageStore support

Huge thanks to all of the contributors including @corbt @rpopovici @dhrrgn @niborb @cdimascio @sunnylqm ! I apologize if I left anyone's name off this list, github autocomplete is not being very helpful today.

Please weigh in with your thoughts. It feels like we have a solid community behind the project now, and if we coordinate efforts a bit we can get to 1.0 in the next month. :balloon:

rt2zz, 2016-04-04:
/cc @lwansbrough

lwansbrough, 2016-04-04:
I wanna send out a huge thanks to everyone who's made contributions to this repository as we move towards a v1. My work hasn't been too deeply involved in React Native lately so it's been difficult for me to find time.

Regarding the proposed changes: the current additions look good. I'm in favour of returning an object for the capture response.. Should have been doing that from the get go. Obviously sub views need to be prioritized for v1. I think barcodes on Android would be a nice to have but for me is the third most important thing there, followed by ImageStore support.

As we continue development I think it's important to highlight a few things:
- Be sure you'd use the code you're writing in production. We don't have tests so extra care should be taken care here and during merge requests. So far this has been pretty good but I've taken part in a couple discussions where a fix for one thing breaks another.
- Look at other modules when looking for naming patterns. Predictability is really important for a good library.
- Where possible, challenge yourself to write for both platforms, or at least be sure that your API would work for both platforms if someone wanted to extend your API.

Again, I'm super appreciative of those who are helping develop this module. Keep up the great work. Hopefully I can get some time at some point to review everything that's changed and get up to date and help out. Until then I'll try to be around more to at least answer questions. Cheers!

chirag04, 2016-05-19:
@rt2zz idk about the roadmap progress but can you publish an 1.0 alpha atleast on npm. Right now rn 0.26 is having a breaking change which is fixed on rn-camera master.

dhrrgn, 2016-05-21:
I think many of those TODOs should/could be consolidated into a single "Android Parity" task.  This could be broken into a deprecate issue with a checklist of items.

As the contributor of the `mirrorImage` prop, I can say I will make an effort to get Android support for this.  When I added that feature our app was iOS only and I had no Android experience.  I now feel comfortable enough with it to support it there as well, so look out for that PR in a few days or so.

chirag04, 2016-08-05:
Not complaining and i appreciate everyones effort in maintaining this package. It's been a long time since the prev release. We should release what we have as is. my guess is most people are maintaining forks or using master branch.

releasing will only help fix bugs faster and get more contributions from the community.

Thanks for the amazing package.

charpeni, 2016-08-28:
@chirag04 You're right.

I will work on a mostly stable `0.4.0` version to be release soon. I just wanted to be sure we have a stable package and a good documentation/README before releasing.

So if you're up to do some triage with issues and pull requests, it will be appreciated.

sibelius, 2017-12-10:
I think we should move this package to an org, something like this one https://github.com/react-native-community

the Android version is using a deprecated api:
check here https://github.com/lwansbrough/react-native-camera/issues/1031
and this one https://github.com/google/cameraview/pull/218

we should use CameraView (https://github.com/google/cameraview) to support all android versions.

the ios version is using a deprecated api as well:
https://github.com/lwansbrough/react-native-camera/issues/465

We can improve a lot of this package using @expo working on Camera package: https://github.com/expo/expo/tree/master/android/expoview/src/main/java/versioned/host/exp/exponent/modules/api/components/camera

lwansbrough, 2017-12-14:
Let me know what I can do help the process of moving this to an organization. I'm more than happy to release my control of this repository if we've got a commitment from some folks to keep things in motion.

sibelius, 2017-12-31:
cameraview fork is here https://github.com/react-native-community/cameraview synced with expo version of it

sibelius, 2018-01-23:
RNCamera is here https://github.com/react-native-community/react-native-camera#experimental

we just need to test more, and we can release 1.0.0

closed 2018-01-23 by sibelius