react-native-camera/react-native-camera #247: Build error

verystrongjoe, 2016-04-04:
I got the following error whenever I build with this libraries.

(null): error: make directory /Users/jouk/Library/Developer/Xcode/DerivedData/mobileOpenAlm-cmeqcfhepzpgbweuaddwjfottcaw/Build/Products/Debug-iphonesimulator/mobileOpenAlm.app/ios: File exists

I tried to work around this with both automatic and manual installations.  But it didn't work yet.

rt2zz, 2016-04-04:
What version of the camera do you have installed? I am not sure when focus square was added, but i have successfully built the latest on master with rn.14

verystrongjoe, 2016-04-04:
Sorry! It was my mistake. I set the build target as a camera project itself. I changed that to my own project. it worked.

closed 2016-04-04 by verystrongjoe