react-native-camera/react-native-camera #249: capture to camera roll crashes on certain Android devices

cdimascio, 2016-04-04:
### Steps to reproduce
1. install react-native-camera on a clean samsung device e.g. galaxy tab 4
2. ensure no photos have been taken
3. use react-native-camera to capture picture and store to camera roll
4. app crashes b/c sdcar/DCIM/Camera is missing

### Expected behaviour
image is captured and saved to camera roll

### Actual behaviour
image is captured and crashes when saving to camera roll because DCIM/Camera directly is not found

### Environment
- **Node.js version**: 5.2
- **React Native version**: 0.3.8
- **React Native platform + platform version**: Android KitKat

### react-native-camera
**Version**: npm version or "master"

dcworldwide, 2016-06-02:
This could be the issue i'm facing (Samsung S5). I'll test later this week to see if switching the photo storage method resolves the issue.

fbacker, 2017-09-29:
Could it be that the camera is 'locked'. 

Try to open the default camera app, does it give an error? restart the device.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco