react-native-camera/react-native-camera #255: playsoundoncap error on android emulator when camera capture ran

Intelurate, 2016-04-11:
### Steps to reproduce
1. Just did a recent pull and started getting this error in the android avd emulator when I ran camera.capture. 
I have not tested this on an actual device. 

[image]

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
ALL LATEST VERSIONS 

### react-native-camera
**Version**: npm version or "master"

rt2zz, 2016-04-11:
Can you reinstall from master, I pushed a [fix for this](https://github.com/lwansbrough/react-native-camera/blob/master/Camera.js#L164) issue a few minutes ago.

Intelurate, 2016-04-11:
Thanks that fixed it.

rt2zz, 2016-04-11:
👍

closed 2016-04-11 by rt2zz