react-native-camera/react-native-camera #257: hello,I add <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>, but I don't see it ?

happyEgg, 2016-04-12:
<Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>

### Environment
- **Node.js version**:v5.6
- **React Native version**:o.22.2
- **React Native platform + platform version**: Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

KevinHu2014, 2016-04-14:
By adding it outside the <Camera> , you'll be able to see it

`<View style={styles.container}>
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={Camera.constants.Aspect.fill}>
        </Camera>
         <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>
      </View>`

KevinHu2014, 2016-04-17:
close

KevinHu2014, 2016-04-18:
closes #257

happyEgg, 2016-04-19:
It's successed!!! ths，
I have a question
this.camera.capture()
      .then((data) => console.log(data))
      .catch(err => console.error(err));

data: content://media/external/images/media/512,
but I get pictures of the actual path is /storage/emulated/0/Pictures/1461061483049.jpg
why?

happyEgg, 2016-04-19:
@KevinHu2014

cubbuk, 2016-04-25:
@happyEgg that's relative path to camera roll, by default camera component save your images to camera roll, if you change the `captureTarget` to `disk` then you will retrieve the original path:

`Camera ref="cam" captureTarget={Camera.constants.CaptureTarget.disk}/>`

happyEgg, 2016-04-27:
@cubbuk you're right!!!!thanks~.~

sibelius, 2018-01-09:
styling issues

closed 2018-01-09 by sibelius