react-native-camera/react-native-camera #258: Warning: Native component for "RNBarcodeScannerView" does not exist

Mactub07, 2016-04-12:
### Steps to reproduce
1. added option onBarCodeRead for IOS;
2. in emulator show "Warning: Native component for "RNBarcodeScannerView" does not exist"

### Expected behaviour
work without warnings in Emulator

### Actual behaviour
"Warning: Native component for "RNBarcodeScannerView" does not exist"

### Environment
- **Node.js version**:
- **React Native version*0.20*:
- **React Native platform + platform version**: iOS 9.3, etc

### react-native-camera
**Version**: npm version or "master"

einicher, 2016-04-22:
i do have this problem with an android phone (galaxy s5 neo) too

derdav3, 2016-07-11:
found a solution? I'm getting the same error

fgrs, 2016-07-25:
same issuue here

marcoacierno, 2016-08-04:
Hmm, maybe you want to post this issue to react-native-barcodescanner instead of react-native-camera? 

(RNBarcodeScannerView is used by react-native-barcodescanner not react-native-camera, or at least I can't find anything related to RNBarcodeScannerView in this project)

https://github.com/ideacreation/react-native-barcodescanner

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco