react-native-camera/react-native-camera #265: Android subviews not showing up on top of camera

Intelurate, 2016-04-22:
Android subviews not showing up on top of camera on device. Tested with Nexus 5. 
render(){
        return (
            <View style={styles.camera_base}>                                

                <Camera
                  ref={(cam) => {this.camera = cam}}
                  style={styles.preview}
                  aspect={Camera.constants.Aspect.fit}
                  orientation={Camera.constants.Orientation.portrait}
                  captureTarget={Camera.constants.CaptureTarget.memory}>
                        <Text onPress={this.cancelClick} style={{position:'absolute', borderWidth: 4, borderColor:'green'}}>Back</Text>
                        {captureSection}
                </Camera>

            </View>              
        );
}

const styles = StyleSheet.create({
  container: {
    flex: 1
  },
  preview: {
    position: 'absolute',
    top:0,
    left:0,
    height: Dimensions.get('window').height,
    width: Dimensions.get('window').width,
    margin:0
  },
  capture:{
    position: 'absolute',
    bottom: 80,
    left: (Dimensions.get('window').width/2)-40    
  },
  camera_base: {
    position: 'absolute',
    top:0,
    left:0    
  }
});

Intelurate, 2016-04-22:
I just pulled them out of the camera so they weren't subviews.

closed 2016-04-22 by Intelurate

thomasjonas, 2016-05-02:
Same problem here... Would be nice if this can be fixed.

Intelurate, 2016-05-02:
I had to pull them out of the camera and absolute position them. Here is what I did to position them. 

            <View style={styles.camera_base}>                                
                <Camera
                  ref={(cam) => {this.camera = cam}}
                  style={[styles.preview]}
                  aspect={Camera.constants.Aspect.fill}
                  orientation={Camera.constants.Orientation.portrait}
                  captureTarget={Camera.constants.CaptureTarget.memory}>                      
                </Camera>         

                <Text onPress={this.cancelClick} style={{position:'absolute', borderWidth: 4, borderColor:'green'}}>Back</Text>
<View style={{ flex: 2, 
                        marginTop: Dimensions.get('window').height/3,
                        marginLeft: (Dimensions.get('window').width/2)-60,
                        flexDirection:'row',
                        position: 'absolute', 
                        height: Dimensions.get('window').height, 
                        width: Dimensions.get('window').width }}>
                    <Image
                      resizeMode={'cover'}
                      style={{ position: 'absolute', height: 120 }}
                      source={ require('../../img/balls.gif') } />
                  </View>
          </View>