react-native-camera/react-native-camera #272: Cannot read property 'Aspect' of undefined

mkhanal, 2016-04-27:
### Steps to reproduce
1. First import of the library throws error

### Expected behaviour
Should instantiate camera

### Actual behaviour
Cannot read property 'Aspect' of undefined

### Environment
- **Node.js version**: v4.4.0
- **React Native version**: react-native-cli: 0.2.0, react-native: 0.24.1
- **React Native platform + platform version**: iPhone6 - iOS 9.2

### react-native-camera
**Version**: "master"

mkhanal, 2016-04-27:
Initial debugging shows that CameraModule is undefined after line 12 in Camera.js

const CameraManager = NativeModules.CameraManager || NativeModules.CameraModule;

-> console.log(CameraManager) here shows that CameraManager is undefined

Dave2011, 2016-05-03:
Hi, I get the same error, why is this issue closed? Is it already fixed?

VansonLeung, 2016-05-06:
If you are using:
var Camera = require('react-native-camera')

Try 
import Camera from 'react-native-camera'

mkhanal, 2016-05-08:
I used import and i got this error

tkjone, 2016-05-10:
@mkhanal I ran into this issue as well.  For me it was because it was not properly installed properly in my xcodeproject.  I ran through the install instructions again and it worked :)

letthewookieewin, 2016-05-10:
Thank you @tkjone, followed the manual installation instructions (no rnpm) and it worked!

mkhanal, 2016-05-11:
@tkjone...unfortunately i am not using xcodeproject. I just use the command line and that's where i hit the issue. I've run through the instructions multiple times and even cleaned and reinstalled the whole environment. :(

Dave2011, 2016-05-11:
@VansonLeung switching from var ... = require to the import worked for me, thank you!

mkhanal, 2016-05-11:
BTW...i'm using it for android development

martinezguillaume, 2016-06-03:
After linked the library into your android project, you need to reinstall the project on your phone/emulator with ```react-native run-android``` for example

zyadsherif, 2016-06-14:
@martinezguillaume that solves the issue, thanks

jqn, 2016-07-13:
I was experiencing the same error in android - the best way to deal with this is to make sure the project is not running when doing the initial install - for ex: stop the android emulator and the android packager running in the terminal.

jqn, 2016-07-13:
In addition to the above the documentation is also inaccurate for react native 0.29.0 - don't make changes to MainActivity.java. Instead make changes to MainApplication.java:
`import com.lwansbrough.RCTCamera.*;`
and

`new RCTCameraPackage()`

commit-master, 2016-10-09:
Need help, I got the same issue on iOS.

leonacky, 2016-11-17:
Android is worked but get error on IOS, I have tried auto linked and manual linked the package
still get error

jjhesk, 2016-11-18:
```javascript

import React, {Component} from 'react';
import {
  AppRegistry,
  View,
  StyleSheet,
  Dimensions,
  TouchableHighlight,
  Image,
  Text,
} from 'react-native';
import Camera from 'react-native-camera';

const styles = StyleSheet.create({
  container: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
    backgroundColor: '#000',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
    height: Dimensions.get('window').height,
    width: Dimensions.get('window').width
  },
  capture: {
    width: 70,
    height: 70,
    borderRadius: 35,
    borderWidth: 5,
    borderColor: '#FFF',
    marginBottom: 15,
  },
  cancel: {
    position: 'absolute',
    right: 20,
    top: 20,
    backgroundColor: 'transparent',
    color: '#FFF',
    fontWeight: '600',
    fontSize: 17,
  }
});

class CameraRoute extends Component {
  constructor(props) {
    super(props);

    this.state = {
      path: null,
    };
  }

  takePicture() {
    this.camera.capture()
      .then((data) => {
        console.log(data);
        this.setState({path: data.path})
      })
      .catch(err => console.error(err));
  }

  renderCamera() {
    return (

      <Camera
        ref={(cam) => {
          this.camera = cam;
        }}
        style={styles.preview}
        aspect={Camera.constants.Aspect.fill}
        captureTarget={Camera.constants.CaptureTarget.disk}
      >
        <TouchableHighlight
          style={styles.capture}
          onPress={this.takePicture.bind(this)}
          underlayColor="rgba(255, 255, 255, 0.5)"
        >
          <View />
        </TouchableHighlight>
      </Camera>
    );
  }

  renderImage() {
    return (
      <View>
        <Image
          source={{uri: this.state.path}}
          style={styles.preview}
        />
        <Text
          style={styles.cancel}
          onPress={() => this.setState({path: null})}
        >Cancel
        </Text>
      </View>
    );
  }

  render() {
    return (
      <View style={styles.container}>
        {this.state.path ? this.renderImage() : this.renderCamera()}
      </View>
    );
  }
};

export default CameraRoute;

```

CameraManager.Aspect is undefined.

ktsein, 2016-11-25:
I had the same problem with iOS version. JPN is right. 
Make sure the project is not running during the initial install to avoid this issue. 
Mine is working now.

hybrisCole, 2016-11-28:
I'm having the same issue when trying to jest snapshot a component using it:

```
test('BarcodeScanner', () => {
  const tree = ReactTestRenderer.create(
    <BarcodeScanner
        onPress={ () => {} }
    />
  ).toJSON();
  expect(tree).toMatchSnapshot();
});
```

result: 
```
 FAIL  js/components/common/__tests__/Components.test.js
  ● Test suite failed to run

    TypeError: Cannot read property 'Aspect' of undefined

      at Object.<anonymous> (node_modules/react-native-camera/index.js:250:78)
      at Object.<anonymous> (js/components/common/__tests__/Components.test.js:3:1)
```

thomasobrien99, 2016-11-29:
@hybrisCole did you have any luck with solving that issue?

hybrisCole, 2016-11-29:
@thomasobrien99 sorry man, I made a post w my use case here so the guys are aware but haven't tried anything else to solve it

jayhuang75, 2016-12-02:
@hybrisCole  Same here! any luck?

YouHan26, 2016-12-28:
IOS:
auto link :  Aspect is undefined error
manual link :  simulator stop when enter the page using Camera
Anybody can save me ? troubled me very.

YouHan26, 2016-12-29:
Update:
IOS manual link now can work well, before I write the usage, after check the Xcode output and change the wrong usage, it worked!

nikolal, 2017-01-06:
I have the same issue, tried auto and manual install.

mantap123, 2017-01-10:
The camera run smoothly on iOS and Android emulator.
But there is error when I run jest test:
`npm test`

```
  ● Test suite failed to run

    TypeError: Cannot read property 'Aspect' of undefined
      
      at Object.<anonymous> (node_modules/react-native-camera/index.js:250:78)
      at Object.<anonymous> (src/views/VideoRecorder/VideoRecorder.js:7:24)
      at Object.<anonymous> (src/views/VideoRecorder/index.js:1:166)

```

sibelius, 2017-01-11:
u can create a mock for react-native-camera

something like this:

```
__mocks__
   react-native-camera.js
```

// react-native-camera.js
```jsx
const Camera = {}
export default Camera;
```

Ajackster, 2017-01-11:
I got this error on iOS and rebuilding on device fixed the issue.

MaxwellGover, 2017-01-14:
Is there a consensus solution for this issue? I've tried pretty much all of the above. I've tried building project in XCode, quitting Simulator and terminal and running `react-native run-ios` again. Still getting the error. I'm running XCode Version 8.1 (8B62). I don't have an actual device I can rebuild on at the moment so that isn't an option.

johanhaleby, 2017-01-20:
I have the same problem with IOS. Started the project from scratch and using the example code presented on the main page.

AlisonWonderlandApps, 2017-01-24:
same issue. I use atom as my environment. I have cleaned and reinstalled node_modules and re-run the project. Checked I have linked elements and put the ios10 warnings in info.plist. I use the import keyword.
Still get this error. :(

aizhaodangning, 2017-03-09:
i got the same error on iOS, need help !

undefined is not an object(evaluating 'CameraManager.Aspect')

MaxwellGover, 2017-03-21:
Like @Ajackster mentioned, for iOS, removing the app from my device and then rebuilding via XCode worked for me.

AshokICreate, 2017-04-26:
can somebody help me to resolve this issue. i am getting this error when i run `npm test`.

SirNeuman, 2017-05-03:
I'm also getting this error when running npm test

SirNeuman, 2017-05-03:
@AshokICreate did you see sibelius's comment for using mocks?

jamietj, 2017-05-09:
I get this error when using react-native-camera with the react-native-navigation package. I have tried building on an Android device (as well as emulator), reinstalling package, deleting node_modules and npm install, update. I have had this camera package working many times but each time I try to get it working with some navigator something breaks. With the original navigator package, it works but each time I go back to the camera the app crashes.
Has anyone got this working with a navigator, if so which one?

fgrs, 2017-06-11:
same issue and I'm using react-native-navigator too :/

mateorecoba, 2017-07-03:
Same issue on ios.
Any updates?

tuckerjt07, 2017-07-27:
I am getting the issue on android. Have performed the manual installation steps as well.

kjkta, 2017-07-28:
- Stop the simulator and build processes
- Run `react-native link react-native-camera`
- Run `react-native ios` | `react-native android`

Seems like a required step. Should be in Getting Started docs?

aaronbuchanan, 2017-08-08:
yes, @kjkta pretty much every native module will need this step, fyi!

stevenleeg, 2017-08-12:
Seeing this issue as well (also using `react-navigation`). Any updates here?

jseverinson, 2017-08-30:
I had this issue in the simulator. Removing the app from the simulator and running `react-native run-ios` again to rebuild fixed it.

victorbadila, 2017-09-29:
+ 1
I did as @jseverinson said and it did not work for me on the iOS simulator. Oddly enough it works well on actual Android and iOS devices so far.

hakanozen, 2017-11-15:
**+1**
any updates?

lightning-zgc, 2017-11-15:
`react-native link`  
May fix that.

sibelius, 2017-12-09:
`react-native link react-native-camera`

rebuilds

closed 2017-12-09 by sibelius

hampelm, 2017-12-13:
I was seeing a similar problem when running jest tests, and mocking `Camera` by adding this at the top tests [per the jest docs](https://facebook.github.io/jest/docs/en/tutorial-react-native.html#mock-native-modules-using-jestmock) of solved it:

```
jest.mock('react-native-camera', () => 'Camera')
```

st4allen, 2018-01-03:
I also have this issue. Then I find that I missed two steps in the documentation.

First remember to add permissions in plist file.

```
<key>NSPhotoLibraryAddUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>
<key>NSCameraUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>
<key>NSPhotoLibraryUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>
<key>NSMicrophoneUsageDescription</key>
```
Then remember to add dependency in your podfile.

`pod 'react-native-camera', :path => '../node_modules/react-native-camera'`
and run `pod install`.

For me it works after these two steps. If it still not work, check if you link the library properly.

Hope it helps.

matheusbtrevisan, 2018-07-05:
i already did.
[image]

godoakbrutal, 2018-08-16:
this work for me in iOS
1. remove  folder `build` in folder `ios`
2. `react-native run-ios`

done.

casperstorm, 2019-03-07:
If someone is still facing this issue, then please remember to follow the steps to ensure [Face Detection](https://github.com/react-native-community/react-native-camera#face-detection-or-text-recognition-steps) is removed from the project.

devtas, 2019-04-16:
In iOS:

1. Open Xcode
2. Import `RNCamera.xcodeproj` from `node_modules/react-native-camera/ios/` to Libraries folder inside XCode
3. In Build Phases of your project, select Link Binary With Libraries and add `libRNCamera.a`.

Works fine for me!

studiobrain, 2019-04-16:
@devtas Im actually attempting this now, and I cannot get a build through.  My `RNCamera.xcodeproj` is located in `../node_modules/react-native-camera/ios/RNCamera.xcodeproj` yet the `libRNCamera.a` file is located in ```User/Library/Developer/Xcode/DerivedData/AppName-bzwvfybuowffzfdkbcxoqnjwfzrm/Build/Products/Debug-iphonesimulator/libRNCamera.a```...  If I search for it in the `Link Binary With Libraries` section it isnt found, yet if I drag it in I get build errors.  Did you do anything differently?