react-native-camera/react-native-camera #274: [Proposal] Drop "orientation" and "aspect fit" properties

vespakoen, 2016-04-29:
Hey!

Does anybody actually use these properties?

- **orientation** is something I cannot imagine anyone using?

- **aspect mode: fit** does anyone want the capture to be squished? or would that leave some edges around the preview?

nschoen, 2016-07-02:
Hi,
well I can't say for now, because orientation does not seem to work. The camera only adjusts it's view when I shake the device (device is always in landscape mode, but camera always starts in "portraitUpsideDown" mode, very confusing). But I could image to use "landscapeRight" or "landscapeLeft" when it's working.

nschoen, 2016-07-02:
I fixed the described problem in #333. The auto orientation works really well. Using the property orientation should not necessary for me any more.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco