react-native-camera/react-native-camera #276: Getting photo up to my own servers

sammosampson, 2016-05-03:
Not really an issue, but do you know the best way I can upload the photo data up to my own server. Should I send the base 64 output from the memory mode output (I here this doesn't work well across the RN bridge though) or do I send the file itself up somehow - if so how?

dcworldwide, 2016-05-06:
I have the same question. Could you please post your design decision if you figure out a best practice? Thanks

knowbody, 2016-05-10:
I just posted a little tutorial here: https://medium.com/@knowbody/react-native-image-upload-to-s3-bucket-5220941bfea2#.9l2o8nmyf

dcworldwide, 2016-06-02:
OK looks straightforward thanks. So RCTCameraRoll is not linked by RN by default? Does this apply to android as well?

knowbody, 2016-06-03:
@dcworldwide it should be linked for you, the reason it wasn't linked for me is because I was using cocoapods and I only specified a few in my pod file.

benjick, 2016-07-13:
So how would I upload the file with fetch?

knowbody, 2016-07-13:
@benjick 
see: https://github.com/marcshilling/react-native-image-picker/issues/1 
and also from the react-native repo: https://github.com/facebook/react-native/issues/201

benjick, 2016-07-13:
@knowbody Thanks, but that I already have the URL using `Camera.constants.CaptureTarget.temp`, can I somehow just add that file to a fetch request?

knowbody, 2016-07-13:
see the link to the tutorial I posted above, I show how to upload photos to S3, might be useful.

charpeni, 2016-07-13:
Thank you @knowbody.

I'll close this issue since he answered perfectly with his Medium post.

closed 2016-07-13 by charpeni

hugohow, 2017-03-24:
It works great for amazon s3, but what about other types of server ?
I'm trying to send it to a Rails app, but I have not succeed to do it .. 
Does anyone have an example of an image of the disk, sent to a server ?? Thanks

hugohow, 2017-03-24:
It's ok, I solve it, it helps me a lot : https://github.com/kalbunna/react-native-image-upload-with-rails-api