react-native-camera/react-native-camera #277: Unable to scan QR Code on computer / phone display

Somojojojo, 2016-05-05:
### Steps to reproduce
1. Display a QR Code on a smartphone or computer
2. Try to scan the code with this library
3. Fail

### Expected behaviour
The code should be scanned

### Actual behaviour
The code is not detected.

### Environment
- **Node.js version**: 5.4.1
- **React Native version**: 0.20.0
- **React Native platform + platform version**: iOS 9.1 (iPhone 6)

### react-native-camera
**Version**: "master"

We also tested with a QR Code on paper, to make sure it wasn't reading anything at all. It scanned the QR Code on the paper insanely fast.

closed 2016-05-05 by Somojojojo

frostney, 2016-05-08:
@Somojojojo Did you find a solution for this? I've been running into the same issue.

Somojojojo, 2016-05-10:
Yeah it was my own stupid mistake for not noticing the issue at hand. The QR Code library I was using was generating the foreground and background colors in reverse, so the QR Code was essentially inverted.