react-native-camera/react-native-camera #278: Get Base64 image

timotius02, 2016-05-08:
How do we actually get the base64 image that we took?

I've tried playing around with the captureTarget, but I was hoping that there was a simple way to just get image.

Thanks in advance

stoffern, 2016-05-16:
You could use "react-native-fs" to read the file into base64 until the Base64 will work 100%.

andrerfneves, 2016-06-23:
@timotius02, make sure you have memory as your captureTarget

`captureTarget: Camera.constants.CaptureTarget.memory`

Then take the picture:
```_takePicture() {
    this.refs.camera.capture()
      .then(data => {
        console.log(data);
        Actions.preview(data);
      })
      .catch(err => console.error(err));
  }```

And then to view it do something like this:
`<Image
          source={{
            isStatic: true,
            uri: `data:image/jpeg;base64,${this.props.data}`,
          }}
          style={styles.image}
        />`

charpeni, 2016-06-23:
`Camera.constants.CaptureTarget.memory` is deprecated.

The best solution would be to read the image as base64 with `react-native-fs` from the path returned by the `captureSuccess` callback. After that, you can display the image with : 
```js
const base64image = await RNFS.readFile(photoPath, 'base64');
```
 ```jsx
<Image source={{ uri: `data:image/jpeg;base64,${base64image}` }} />
```

closed 2016-06-23 by charpeni

stoffern, 2016-06-24:
That is actually not correct, it would be like this:
```js
import RNFS from 'react-native-fs'

RNFS.readFile(file.substring(7), "base64")  //substring(7) -> to remove the file://
.then(res => this.setState({base64: res}))  //res will contain base64 with data:/image/*:base64;..

<Image source={{ uri: this.state.base64 }} />
```

mtwallet, 2016-06-25:
@stoffern I wonder if you can help me? I am trying to implement your suggestion above for iOS but I am getting the following error:

`2016-06-25 15:32:28.350 VehicleCheck[2438:1856471] -[<CALayer: 0x12df60ba0> display]: Ignoring bogus layer size (288.000000, 238119.000000), contentsScale 2.000000, backing store size (576.000000, 476238.000000)
2016-06-25 15:34:06.245 [info]`

I have `captureTarget` set to cameraRoll (although it doesn't seem to matter which setting I apply) `captureQuality` is low and my code looks like this:

`captureImage() {
  if( this.camera ) {
    this.camera.capture()
      .then( ( data ) => {
        RNFS.readFile( data.path, 'base64' )
          .then( res => this.setState( { uri: res } ) )
      });
  }
}`

If I log the output of `data.path` I get:

`/var/mobile/Containers/Data/Application/8071BF6D-4705-4A02-8746-4C7AC5BFE99E/Documents/6BB91BF9-BEEA-44C3-B823-38506D4FFCBB.jpg`

Am I doing this correctly?

stoffern, 2016-06-25:
@mtwallet To me it looks like a informational dump, not a error `[info]`, anyways this is related to native-camera capture size and scaling and does not give us anything.

I have not tested it on iPhone, but what output do you get if you run this command:

```js
var file = '/var/mobile/Containers/Data/Application/8071BF6D-4705-4A02-8746-4C7AC5BFE99E/Documents/6BB91BF9-BEEA-44C3-B823-38506D4FFCBB.jpg'
RNFS.readFile(file, 'base64')
.then(res => { 
  console.log('read ok')
  console.log(res)
})
.catch(err => {
  console.log('read error')
  console.log(err)
})
```

mtwallet, 2016-06-25:
@stoffern Just tried and I get:

`/9j/4AAQSkZJRgABAQAASABIAAD/4QVuRXhpZgAATU0AKgAAAAgABAEaAAUAAAABAAAAPgEbAAUAAAABAAAARgEoAAMAAAABAAIAAIdpAAQAAAABAAAATgAAAAAAAABIAAAAAQAAAEgAAAABAByCmgAFAAAAAQAAAaSCnQAFAAAAAQAAAayIIgADAAAAAQACAACIJwADAAAAAQGQAACQAAAHAAAABDAyMjGRAQAHAAAABAECAwCSAQAKAAAAAQAAAbSSAgAFAAAAAQAAAbySAwAKAAAAAQAAAcSSBAAKAAAAAQAAAcySBwADAAAAAQAFAACSCQADAAAAAQAQAACSCgAFAAAAAQAAAdSSFAADAAAABAAAAdySfAAHAAADOgAAAeSgAAAHAAAABDAxMDCgAQADAAAAAQABAACgAgAEAAAAAQAAAJCgAwAEAAAAAQAAAMCiFwADAAAAAQACAACjAQAHAAAAAQEAAACkAgADAAAAAQAAAACkAwADAAAAAQAAAACkBQADAAAAAQArAACkBgADAAAAAQAAAACkMgAFAAAABAAABR6kMwACAAAABgAABT6kNAACAAAAIgAABUQAAAAAAAAAAQAAADIAAAALAAAABQAAHMEAAAUYAAAe1AAADY0AAAPzAAADawAAAAAAAAABAAAAUwAAABQAXwBHAIwAP0FwcGxlIGlPUwAAAU1NAAoAAQAJAAAAAQAAAAQAAgAHAAACLgAAAIwAAwAHAAAAaAAAAroABAAJAAAAAQAAAAEABQAJAAAAAQAAAJ0ABgAJAAAAAQAAAJUABwAJAAAAAQAAAAEACAAKAAAAAwAAAyIADgAJAAAAAQAAAAAAFAAJAAAAAQAAAAEAAAAAYnBsaXN0MDBPEQIAWgBUAE8ATQBJAEgARgBaAJoAyACqAKQApgB4ACMAEABfAFcAUwBOAEwASABGAJIA1QCdANEA5QB2AGsAEAAPAGUAXABWAFEATgBKAEcAqwCXAOsA2wCzAH0AaAAIAA0AagBhAFwAVQBSAEwASQDFAP8A4QCsAHcAngCTAAgADAByAGgAYgBaAFUAUABPADEB/wCmAKsAuQBxAEwABwAMAHsAcQBpAGEAWwBVAEwALQG/AOcArwCKAIQAQAAHAAwAhQB6AHEAaQBiAFsATgA+AfYAzACaAFMATgBbAAcADACQAIMAegByAGwAZgB4APwArgB0AFYAYgBNADMADQAMAJsAjwCEAH0AeAB2ANYA8wBlAGkAMQAUABQAHQApABMApwCZAI8AiACGAIgAAAGZAWoAFAAUABsAJQAvAE0AkgCzAKUAmwCUAJQAnQBBAd0BDgE3ACEANwBLAIEAzgD0AL8AsAClAJ8AoQCuAI0B8gGaAakATwCdAMQA3gDbAM4AzAC8ALAAqQCtALsA2AEVAsgBBgGLAK0AuACqAJMAhwDXAMgAuQCzALYAxAAEAjYC2AFeAc8AYwBaAFkAUgBVAOMA0gDEALsAvQDKAEICVwLnAZ4B+wCmACwAMgA4AEEA7QDcAMwAwgDBAM0AyAF1AukBvwE9AdIAcwAjACsANQAACAAAAAAAAAIBAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAIMYnBsaXN0MDDUAQIDBAUGBwhVZmxhZ3NVdmFsdWVZdGltZXNjYWxlVWVwb2NoEAETAAEu1rzEVnwSO5rKABAACBEXHSctLzg9AAAAAAAAAQEAAAAAAAAACQAAAAAAAAAAAAAAAAAAAD////lKAADLO///9f0AAA0g///1VwAAEMwAAABTAAAAFAAAAFMAAAAUAAAACwAAAAUAAAALAAAABUFwcGxlAGlQaG9uZSA2IGJhY2sgY2FtZXJhIDQuMTVtbSBmLzIuMgD/7QA4UGhvdG9zaG9wIDMuMAA4QklNBAQAAAAAAAA4QklNBCUAAAAAABDUHYzZjwCyBOmACZjs+EJ+/8AAEQgAwACQAwEiAAIRAQMRAf/EAB8AAAEFAQEBAQEBAAAAAAAAAAABAgMEBQYHCAkKC//EALUQAAIBAwMCBAMFBQQEAAABfQECAwAEEQUSITFBBhNRYQcicRQygZGhCCNCscEVUtHwJDNicoIJChYXGBkaJSYnKCkqNDU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6g4SFhoeIiYqSk5SVlpeYmZqio6Slpqeoqaqys7S1tre4ubrCw8TFxsfIycrS09TV1tfY2drh4uPk5ebn6Onq8fLz9PX29/j5+v/EAB8BAAMBAQEBAQEBAQEAAAAAAAABAgMEBQYHCAkKC//EALURAAIBAgQEAwQHBQQEAAECdwABAgMRBAUhMQYSQVEHYXETIjKBCBRCkaGxwQkjM1LwFWJy0QoWJDThJfEXGBkaJicoKSo1Njc4OTpDREVGR0hJSlNUVVZXWFlaY2RlZmdoaWpzdHV2d3h5eoKDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uLj5OXm5+jp6vLz9PX29/j5+v/bAEMAAgICAgICAwICAwUDAwMFBgUFBQUGCAYGBgYGCAoICAgICAgKCgoKCgoKCgwMDAwMDA4ODg4ODw8PDw8PDw8PD//bAEMBAgICBAQEBwQEBxALCQsQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEP/dAAQACf/aAAwDAQACEQMRAD8A96ll5xms+SQZ5pryetUZX5zmvk2j34iyyVnTSgDmiSU/Ss6aT3qLmiRHLIc4zWbLMSMA81NK27JrKlkHQVLZsitNK2eDWVPK33SeKuSk5ye9Y07sOlLmLK0zjpWLOwDE5NXp5DyO9Ys8uaaYNFWZmxkHNY8zbDnrmrs0oGTjArHkmBySeDWkUQ0VJpuSazJGDfODk1cmYMSM1lTYHyjkEVokIilmLkGQY28cVnSOueBxSu5yRjNVnLA4xnNaJEyGyFSB61AZCG9hTXYk9DURf+GmUf/Q9YkkwOTVF5c59ulRSyZGM1SaQdPSvlGfQRQryHpmqMrjOOtPdyOepPaqUj54HWsmzeKIZHGDzWdM+OoqeZjms6V+MZqLmqRTml4rKuJMjNXJW4JzWRPLnNCLsUbhz161kTyHIK8AdauzSHnjJxWLNIOQOPxqkS0VJpPmOazJXHQ9KmmducnismZyeP51skSyK4kIJxzWY8p6nPFOnYjOOcVQlk5yeeOlbRRmxHkUr0qizncSOc0O5wAPqaqGRuxqoxIHO5zk9qhLgHNDOecmoGJ3ZBqyuY//0e3dzVZmwaczEioyTivkZyPpVEiYDOaqOBg4q0wY4Oahdc8VhKRtGJmSe9Z0xHIrUmU5/wAKzpo85JqFI1ijGuMD5qxZznrW3OG/KsK5B57VUZFtGRM3BNYdw/zEkYrVuevBxmsG4Yg5PNaxMWjPmkHXpWVKx5qzct3Ix71kzPheT+VbxM2QSOvIB61RlXP8WBTpWDcVSY/WtjNsa+T3zVZwduO4pTk8VCVI5JrVIjmAuG+9UJfJJzkU1ic8dqTbkdQKTRSZ/9LstvGKbsOatCM9ak8rua+JnM+sjEobMc1BItanlcYPWoTF1rByNoxMWSPtWbNG2D6V0EkfWs2eI9O9ZuZsoHLXCVzl1x9RXX3cWAc1yt6u0VcJCcTlrpsHBrnLl/m9q37w4ya5O8ccnNdlN3OeRnztnv8AnWTMwJwD0qS5esuV29eldcEc0mNkkGDjgCqpm60x2BFVmIwTmt0jJsdJJu6nFV2lJ6mmk9hUbHNaCJPu96QOVOeOKjLYqLJxihoVz//T7d7+KEKZIZBn/YNOGraccBnKH/aUiusgdnZIkUSMeAMZJNfXPwp/Z/i1QQ+I/G1oqxcPFasvLehf0HtX5Xl+MrYifIo3P1HM8FhsPD2kpWPl7S/h94w1zTU1jSdGuru0k+7LHESpx6VmX3g/xLYk/bNKuocf3oXH9K/YO3t7Swt47e2jWCGIBURQAoA4AAFUtQvkhQxxQi7usAiAMocg9/mOAK+ullSt8R8Ys2d/h0PxjnsbiM4kjZD6EEVlT25zyK/au50HQbyP/TtPt5Cw5EkSN/SvN7/wH8MdXZi3h+xmhUkPIIwnPTA24ya8+plklszso5qn0Z+QF1b5BxXH6jAQDkV+sPif4JfCEWzXX9kNGG4/cSuoXPc8nArxPVP2cPBOpXi21hcXMCSoX3pIsoTGMA5GTmphgKvRHR9epdWfmZqGVJPSuIvpASa/SLxD+x7ZMGOm+JGWRj8qSQg/qCOlfIvx1+Cr/BjRDr2s+I7O4Vm2pAqsszepAORgDk81208JUW6OOpioN6M+dJpOvtWVLIc81ykPjzQ71Q0U2QfUVow6rDfkC1DSE9lQn+QrsjRkt0YOtF9S8zgj0qqW7dq1V0PxBLGJE0q6Kn+IQSY/PbWbcWt7AcXFvJEw67kZf5irSM1UXcrlscEVGW9DTGkHfrQGHNaBckyMY70m7ioy46igDPIqGhn/1P1b+EnwC0zwiIdb8Sqt5quAyoeY4T7Du3vX0bc30VtEWyCR2HPSufutVmhjYopWRuc9cA9M1xOp60lwyaPHqKW+oXW4qCpLDaPm2kcZHXrXyWFVOjHkpqx9BiFVxE/aVmdfqGr3cSAWdtJczXA+UoBiMY6uCQB9M81E9zY6PF9tvP3tzj5pSMuWb+EdSBnsDgVztnNcaTpqw314bqUKACuE3EDGAPfqcmuZlvXv5JMTKxJw4GD5fqoI7+9ae1bYvqyR1M+uX1/K0RVo45FG5g33Qf4QO59TWRc6xaaTbSSTMkMMQJOM8AcnisfUNTg0i1e5uZvLhhAHoPZR6k9KwLKK48QXP9tyqZNOuYgRDPGA4PGMZJGOua1jLqwcbaIvza7qt1a/arJLe6ikAwscpDEnp8pB6ikllXSbeNryYblxnGASx/hAAFP1N9M0WOTWHhSH7PH95FCttUfdGP0r5w1/xuNUuF8QXIuLOSKNhHFM4Kxp3cheNxHr0p1cZGmuZhSwkqjsjt/iF8UNK8M2FzqVxMY3jXLDg49FAweT+tfPPh7wH4Q8fzHxp8XZIdUl1AAwWM8aslpCxO1fncAE9WOK+Ff2n/il4h1jTHuvDcpj02zmG+bk7ucbyfTPSue8E/Cvxn4t0+21HUvG06QXKBgsMDuSDxw0joDXZgcPjaz9pThdG0qmVwg44mrb0v8Aofq1q/hH4GeDPDWpa34M8L6TqM+k2M9ybZILcMzwqW25AOM4614t4B/bd/Zwm0K2v/EEelaLNLbB2ggT99FNjlCNuOKxvhd8OPDXwwsb1bjVbvV9Q1ZRaFrgoEjEmAQUVm4ORzmvnL4d/B34IX/xv8TfC/xl4Ogi1DTxHcW7PIypMkwLEhV7ZIr67A0q8bU61NczR+ [... cut ...] Pp2PavSozVjz6sTT8VaWt1pro6FQARu5wf97BNfm58RdJ+xapNE67Pmbb7/AEr9RnSWS2ZZSURhwORgevviviP4z+HDHdSTAZLZ5xnP+Fc2NWlzXCb2Ph69tG3sCvBziuC8TRCzszcnh1I2EHB3e1ex6jZtC0hkYLGmS2TjGO9fOHifUv7V1R2ifdbx/LHjoR61hl6c5+SOnHyUYa7slPjrxEybZLln7ZOc8VRn8V6zN1nK59KxDDz60vle9e37GHY+f9p5lt9a1R/vXD8/7VVJLu6l+9IzfUk0oj46U5Yq0UY9hXbIAzk5PWu/8F32j22oC41K3WVolLRo2Npk/hznqAeoriFj9akWPn0qKkbqxvhq8qVRVIq7XfU+grfwfdaur3+6P98SxK+/sOlXofhnIw3tG8g9hgV5p4U8e6z4YlUxv50QIyp649Pevrrwj8S/B3im0W2d/sl3/Ej4AJ9jXz+YV69FXiro7cHQpVZN1JWZ/9k=`

mtwallet, 2016-06-25:
I've adjusted my code to the following and I am now getting the base64 string as I expected:

`captureImage() {
    if( this.camera ) {
      this.camera.capture()
        .then( ( data ) => {
          let base64Img = data.path;

          RNFS.readFile( base64Img, 'base64' )
            .then( res => this.setState( { uri: res } ) )
        });
    }
  }`

@stoffern I really appreciate the help there!

charpeni, 2016-06-25:
@mtwallet like I said. :)

stoffern, 2016-06-25:
@mtwallet np ;)

pacozaa, 2018-01-24:
This is weird why #892 point to the direction that using `data.data` is fine but this is even an older issue but suggest `react-native-fs` which i guess it's more convincing because `data.data` is `undefined`. Yesterday, I also create new issue #1149 

Why is this not mentioned and correct in README.md?

Anyway thanks everyone. I am gonna bet for using `react-native-fs` solution.

sundariragavan, 2018-11-13:
Now i'm using "react-native-fs" for image save to particular folder.
The folder is created and image also saved in destination path but the image is not displayed in **Gallery**.
How to fix the issue.

```const path = `${RNFS.PicturesDirectoryPath}/My Album`;
		await RNFS.mkdir(path);
		return await fetch(uri)
			.then(res => res.blob())
			.then(image => {
				RNFetchBlob.fs.readFile(uri, "base64").then(data => {
					RNFS.appendFile(`${path}/${image.data.name}`, data, "base64").catch(
						err => {
							console.log("error writing to android storage :", err);
						}
					);
				});
			});```

n1ru4l, 2018-11-14:
@sundariragavan You will have to use CameraRoll api (https://facebook.github.io/react-native/docs/cameraroll.html) to save an image to the camera roll.

n1ru4l, 2018-11-15:
@sundariragavan Probably my moving the picture there with react-native-fs and then using the destination file uri with as the first param of `CameraRoll.saveToCameraRoll`?

sundariragavan, 2018-11-16:
@n1ru4l  If i'm use <CameraRoll.saveToCameraRoll>the image is saved in ***DCIM*** folder.

elirangoshen, 2021-06-17:
i still cannot use it as i'm getting an error : 
Error: ENOENT: no such file or directory, open 'ED7AC36B-A150-4C38-BB8C-B6D696F4F2ED/L0/001'

im trying from iphone..

felipevyg, 2021-09-08:
@elirangoshen Did you find a solution for this one?