react-native-camera/react-native-camera #279: Is there a onBarCodeRead method supported in Android ?

blacksunset, 2016-05-08:
Hi, I am building a app with QR code reading function.  It seems the OnBarCodeRead function does not work in android, is there any solution  to solve this problem?

xiaomilii, 2016-05-10:
i have the same qustion, Will you solve it?@blacksunset

blacksunset, 2016-05-11:
@xiaomilii i solved it by changing a component.  @see react-native-barcode-scanner-universal

jrm2k6, 2016-05-24:
Same question here, would be nice to be able to use only one library for both platforms

ThaJay, 2016-07-08:
+1
react-native-barcodescanner is not doing it for me at the moment

ThaJay, 2016-07-11:
well react-native-barcodescanner kind of works but looks like it isn't maintained at the moment. So how should we go about implementing ZXing here? I'm willing to help.

jrm2k6, 2016-07-11:
@ThaJay I used the other package mentioned and it seems to be working fine. I don't know how we could go to merge those two projects together basically.

M1chaelTran, 2016-12-17:
there is QR code support now with the latest version 0.4.0 with PR https://github.com/lwansbrough/react-native-camera/pull/413.

consider closing this issue

Disjes, 2017-03-11:
is BarCode supported on Android or just QR Code?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco