react-native-camera/react-native-camera #280: Video not matching aspect fill size?

getnashty, 2016-05-08:
**I have my Camera styles as such:
**
`flex: 1,
justifyContent: 'flex-end',
alignItems: 'center',
height: 350,
width: Dimensions.get('window').width`

**My camera is:
**
`<View style={styles.cameraContainer}>
  <Camera
    ref="camera"
    style={styles.preview}
    captureMode={Camera.constants.CaptureMode.video}
    orientation={Camera.constants.Orientation.portrait}
    keepAwake={true}
    torchMode={Camera.constants.TorchMode.auto}
    aspect={Camera.constants.Aspect.fill}>
    <Image source={require('../../img/flip_cam.png')} style={styles.flipCam} />
  </Camera>
</View>`

**My record function:
**

`_recordStart() {
  function handleData(data) {
    console.log(data);
  }

  this.refs.camera.capture()
    .then(handleData)
    .catch(err => console.error(err));
}`

My viewfinder/camera seems correct in record mode (size wise), but when I view the video stored in my camera roll, its just full screen (as if my 350 height was totally ignored). 

Not sure what I'm missing?

getnashty, 2016-05-12:
This must be done post processing

closed 2016-05-12 by getnashty

salimbutex, 2017-12-04:
HI dear,
I am having exact same issue. I need to recorde fully square area...
You said, it's about post processing. But I believe, there must have an option to mention recording area.. instead full screen always... In ios we fixed it...