react-native-camera/react-native-camera #283: Don't use RCT class prefix.

nicklockwood, 2016-05-11:
This looks like a great module, but please don't use the RCT prefix for your classes - that's reserved for first party React Native components so that we can avoid conflicts. If we ever add a camera module to React Native core, we'd probably call it RCTCamera too, and then that would break your module.

A better prefix would be something specific to your project, e.g. RNC (React Native Camera), or your name, e.g. LWAN,  which minimises the chance that someone else will accidentally use the same prefix.

(ps, That doesn't apply to the JS wrapper for your module, which we suggest you leave un-prefixed.)
(pps, And yes, it's kinda dumb that Objective-C uses a global namespace, but that's the way it is)

rt2zz, 2016-05-11:
👍 This would make a great PR if anyone is up for it. I would vote for `RN` as a prefix since that seems to be common in the community, although the risk of collision is still non-negligible.

nicklockwood, 2016-05-11:
@rt2zz Being common is the exact opposite of the purpose of the prefix :-)

sibelius, 2017-12-10:
I'll try to work on this one

sibelius, 2018-01-23:
fixed on master with new implementation

RNCamera https://github.com/react-native-community/react-native-camera#experimental

closed 2018-01-23 by sibelius