react-native-camera/react-native-camera #284: neither latest or github version seems to work

mmmurf, 2016-05-12:
I'd been using react-native-camera successfully for a few weeks (the latest github version as of a few weeks ago).  I recently updated to the most recent github version, and now my code gets an odd error:

cannot read property then() of undefined.  

The code triggering this is:  this.camera.capture().then((result) => {...}) which had worked before. 

It seems that if I capture the return value from capture() it is a promise, but when I attempt to resolve it, the same error occurs.  Even explicitly calling Promise.resolve on the output does not seem to work.

Any suggestions would be much appreciated.  Also if anyone knows the most recent version that definitely works, I can always use that for now.

rt2zz, 2016-05-12:
Is it possible you are calling `this.camera.capture()` before the camera ref is assigned? As a simple test you might wrap the capture in a setTimeout

mmmurf, 2016-05-13:
@rt2zz I checked, and the ref is assigned.  The oddity seems to be in the then() call which is peculiar, it seems that whatever capture() is returning is not exactly a promise.

rpopovici, 2016-05-13:
if you are testing on iOS simulator the capture method is broken. I introduced a regression with the latest commit. but if you are testing on a device it should work.

mmmurf, 2016-05-13:
@rpopovici ahh ok, I hadn't tested on device yet b/c it wasn't working on the simulator. Thanks for the info.

mmmurf, 2016-05-13:
@rpopovici indeed it does seem to work as expected on the device.  I'm curious why it wouldn't work on the simulator.

pyrabbit, 2016-05-15:
@rpopovici not sure if this is related to the regression but the following is stopping me from capturing `[RCTCameraManager.m:242] Error setting property 'captureAudio' of RCTCamera with tag #7: capturing audio` ... Is this related to the regression you were talking about?

rpopovici, 2016-05-16:
@pyrabbit  don't think so

lvelours, 2016-05-19:
Same as you @pyrabbit, i've just updated to the last version of component, and this morning just updated too my phone to ios 9.3. I'm stuck, don't know what's going on...

bjacog, 2016-10-19:
@pyrabbit did you manage to find out what caused the issue?

pyrabbit, 2016-10-19:
@bjacog Unfortunately I did not. I ended up writing a straight Swift implementation of my app though so I didn't put much thought into it. Good luck!

sibelius, 2017-12-08:
check if your package is linked properly on native side

closed 2017-12-08 by sibelius