react-native-camera/react-native-camera #285: react-native-camera capture button on android does not work inside of camera component

lorrainefox, 2016-05-12:
### Steps to reproduce
1. Setup react-native-camera according to instructions on js coach site; use provided sample code
2. Run on ios emulator and read device - works as described, I can take pictures with "Capture" button
3. Run on Android emulator and read device and Capture button does not appear -> Move following line to outside of Camera component and the Capture button appears.
Expected:
The code would not need to be modified to work on Android.  Is there something I need to do to get this to work without varying the source code?

lorrainefox, 2016-05-12:
To be more exact, I moved this line of code outside of the Camera component to get it to work on android:  <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>

However, when I move that line to make the camera work in android, then it no longer works in ios.

stoffern, 2016-05-16:
I have added a comment on how to get the button outside camera here:
[https://github.com/lwansbrough/react-native-camera/issues/198#issuecomment-200468857](url)

closed 2016-06-13 by charpeni