react-native-camera/react-native-camera #286: subview doesn't work for me

rongzhou, 2016-05-15:
### Steps to reproduce
1. use the code from master.
2. copy the test code from readme.
3. test on genymotion android emulator run on mac os.

### Expected behaviour
See the capture button

### Actual behaviour
Can't see the capture button

### Environment
- **Node.js version**: 6.1.0
- **React Native version**: .0.25
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc Android 5.0

### react-native-camera
**Version**: npm version or "master" master

stoffern, 2016-05-16:
duplicate issue..
I have added a comment on how to get the button outside camera here:
[https://github.com/lwansbrough/react-native-camera/issues/198#issuecomment-200468857](url)

gchiocchio, 2016-06-01:
it seems an z-order problem in class RTCCameraView, i have added this code in RCTCameraView.java

```
import android.view.View;
.....
  @Override
    public void onViewAdded(View child) {
        if (this._viewFinder == child) return;
         this.removeView(this._viewFinder);
        this.addView(this._viewFinder, 0);
    }
```

and for my project sub item now work correctly

rt2zz, 2016-06-01:
@gchiocchio good find, can you PR that change?

gchiocchio, 2016-06-01:
yes i have sent the pr

rt2zz, 2016-06-01:
I see [your branch](https://github.com/gchiocchio/react-native-camera/tree/patch-1) but no PR...

cyphire, 2016-07-01:
Thank you gchiocchio! My app works now!

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco