react-native-camera/react-native-camera #290: Add overlay to camera

MathiasLund, 2016-05-19:
Hi there,

Great repo! 
This is not really an issue, just a question:

Is it possible to add an overlay to the camera?

rt2zz, 2016-05-19:
@MathiasLund you can do an overlay using normal react-native styling e.g.
```js
<Camera />
<View style={{position: 'absolute', top: 0, left: 0, right: 0, bottom: 0}}>
</View>
```

charpeni, 2016-06-13:
You can also check the exemple in the PR #304.

closed 2016-06-13 by charpeni

cwcorbin, 2017-07-25:
Is it possible to add multiple overlays, and then capture the video and overlays into a new video?  (That way when it is played in the future, the overlays are a part of the video.)

PatchedPixels, 2017-08-14:
I am also for looking for the same solution, How to record video with overlay view using react-native, A prompt help would be really appreciated. Thanks

storerjeremy, 2017-09-15:
@cwcorbin @PatchedPixels Did you guys ever figure out how to record video with an overlay?

PatchedPixels, 2017-09-15:
We did some prototype work and have idea about how to do it exactly, we now need someone to fund this project for us and we will be more than happy to contribute, we may work even without someone funding but it will be slow that way.

PatchedPixels, 2017-09-15:
The high level idea is to render using OpenGL when recording the video, we need to render the video by using OpenGL & any overlay objects that are in scene. This needs to be done for Andriod as well as iOS and then those functions can be called through react.

PatchedPixels, 2017-09-15:
@storerjeremy ^^

josegoval, 2021-07-21:
> The high level idea is to render using OpenGL when recording the video, we need to render the video by using OpenGL & any overlay objects that are in scene. This needs to be done for Andriod as well as iOS and then those functions can be called through react.

Any real implementation of this?

hakeemsyd, 2021-11-29:
Any example on how to do this?