react-native-camera/react-native-camera #291: Generating a video thumbnail?

getnashty, 2016-05-20:
Does anyone have ideas for how to generate a thumbnail of the video being recorded?

I'm trying to Camera.constants.CaptureMode.still as an option into the capture() instance method (as such `capture([Camera.constants.CaptureMode.still])`, but nothing is happening (the docs specify that it takes options, but also says the arg is just callback, so I'm not clear on this point).

Any ideas?

getnashty, 2016-05-20:
Nevermind, seems like the docs specify capture([options]), but should be capture({options})...

So capture({mode: Camera.constants.CaptureMode.still}) I believe is working (but I haven't finished yet).

closed 2016-05-22 by getnashty

bsiddiqui, 2016-08-26:
@getnashty were you able to figure out how to get a thumbnail from the captured video? From the docs it seems like if you pass mode to options the camera will capture a still instead of a video.

getnashty, 2016-08-26:
hi @bsiddiqui - I did figure this out. The way I did it, was unfortunately hack-y. But because there is no video encoding going on before it hits the server, I had no other way of making a thumbnail. 

So when someone starts recording, the first thing I do is take a picture, then immediately after start recording a video. So the video might miss a moment, but it works for this use-case.

matart15, 2016-11-09:
@getnashty Something like this
```
  startRecording() {
    if (this.camera) {
      this.camera.capture()
        .then((data) => {
          console.log(data)
          this.camera.capture({mode: Camera.constants.CaptureMode.video})
            .then((data) => console.log(data))
            .catch(err => console.error(err));
        })
        .catch(err => console.error(err));
      this.setState({
        isRecording: true
      });
    }
  }
```

What about shutter sounds?

GeoffreyPlitt, 2017-06-11:
Is there a way to extract the thumbnail from a specific timecode in the video?

bz123, 2017-09-07:
@GeoffreyPlitt you have to do that using a service on server side.

GeoffreyPlitt, 2017-09-07:
Nope, we're using react-native-video-processing which does it on the device.