react-native-camera/react-native-camera #293: Crash if app can't connect to the camera

khakulov, 2016-05-24:
### Steps to reproduce
1. App has authorization to access to the camera
2. But camera is busy in hardware level for some unpredictable reason, so that CameraService connection rejected
3. App crashes

### Expected behaviour
check if camera successful connected, if not show and error.

### Actual behaviour
Crash without reason.
`adb logcat` output:

[image]

dcworldwide, 2016-06-02:
@khakulov 

- Is this android specific?
- When does the crash occur? On camera load or on capture?

I have frequent crashes on Android, Samsung S5 - but only when I capture a photo.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco