react-native-camera/react-native-camera #294: Add white balance?

heri, 2016-05-24:
Is there a way to set white balance in `react-native-camera` ?

I'm working on an app where accuracy of colors is key.

One idea if to have the user take a pic of a 18% gray card with which to set the white balance. 

Another alternative is having the user input a white balance (e.g. 4500K, 5000K, 6000K etc.)

Thank you

stoffern, 2016-07-13:
@heri this would be resolved by using `gl-react-native`
you have all kind of filters that you can use:
```
gl-react-temperature
gl-react-sepia
gl-react-sharpen
++++
```

I think react-native-camera should stay as it is a standalone camera component, and camera only.

heri, 2016-07-13:
Thanks @stoffern 

I have researched this and it looks like having direct access to the camera and the raw photography data (such as set temperature) is only possible for the latest iPhone devices (iPhone 6S Plus, iPhone 6S, iPad Pro) and a few android devices. 

`gl-react-temperature` can help but it would be the equivalent of applying a filter (like an orange or blue filter) on top of a picture. It helps but probably not enough for my use case. Which means I cant really use react-native, not in the 1st version at least. 

Thank you !

closed 2016-07-13 by heri

dylanpyle, 2017-02-06:
Any chance we can reopen this one? Looks like [there are APIs](https://developer.apple.com/reference/avfoundation/avcapturedevice/1624568-setwhitebalancemodelockedwithdev) for locking white balance to a certain value.

Seems like that would definitely fall under "a standalone camera component", since it's not the same as adding a filter on the photograph after the fact.