react-native-camera/react-native-camera #296: Random errors due to uncaught exception on iOS

J-Gonzalez, 2016-05-24:
Occasionally, when loading the camera view, I get the following error in logs:
```
2016-05-24 13:33:26.002 myApp[734:897165] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMetadataOutput setMetadataObjectTypes:] - unsupported type found.  Use -availableMetadataObjectTypes.'
*** First throw call stack:
(0x1829c2e38 0x182027f80 0x189153068 0x1001c54c8 0x1003a1a7c 0x1003a1a3c 0x1003ae554 0x1003a572c 0x1003a1a3c 0x1003b0c9c 0x1003b0364 0x182625470 0x182625020)
libc++abi.dylib: terminating with uncaught exception of type NSException
```

The highlighted code is the following:
```
    AVCaptureMetadataOutput *metadataOutput = [[AVCaptureMetadataOutput alloc] init];
    if ([self.session canAddOutput:metadataOutput]) {
      [metadataOutput setMetadataObjectsDelegate:self queue:self.sessionQueue];
      [self.session addOutput:metadataOutput];
  -->    [metadataOutput setMetadataObjectTypes:self.barCodeTypes];
      self.metadataOutput = metadataOutput;
    }
```
Using the following:
  "react": "^0.14.5",
   "react-native": "^0.22.0",
   "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git", (Currently at 0.3.8)

and testing on iOS 9.3.1

I haven't been able to pinpoint the reason for this, as it seems to happen randomly.

Phoenix1708, 2016-08-06:
Same issue here:
"react": "15.1.0",
"react-native": "^0.27.2",
"react-native-camera": latest master branch
iOS 9.3.4

msom, 2016-12-01:
I have the issue when scanning data matrix codes:
"react": "15.4.1",
"react-native": "0.38.0",
"react-native-camera": latest master branch
iOS 10

Using `'datamatrix'` produces the error, `'org.iso.DataMatrix'` (that's what onBarCodeRead's e.type returns).

Possibly related: #347.

Andruschenko, 2017-08-18:
@Phoenix1708 Your sentence somehow stops.. Where do you remove the `barCodeType` property from?

chhornponleu, 2018-01-03:
Any solution on this?

sibelius, 2018-01-03:
Can somebody setup a repo with this issue?

n1ru4l, 2018-09-24:
Closed due to missing feedback.

closed 2018-09-24 by n1ru4l