react-native-camera/react-native-camera #298: Is there a way to save the video clip in MP4 format on iOS?

ranjitpandit, 2016-05-25:
I need the video to be able to be played both on iOS and Android so is there away to save a recorded video clip in MP4 format? Currently on IOS it defaults to .MOV format.

taltultc, 2017-02-01:
I have created a model for that.
https://github.com/taltultc/react-native-mov-to-mp4

zhenghow93, 2017-11-25:
@taltultc thanks for the effort! But would like to know if this will take a long time for large files (say >100mb)?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

arapocket, 2018-03-01:
Someone please open this, as Chrome doesn't support .mov files.

sibelius, 2018-03-01:
@arapocket you should use this https://github.com/taltultc/react-native-mov-to-mp4

@taltultc could u send a PR adding this to the docs of react-native-camera?

arapocket, 2018-03-02:
Thanks that worked fine.

KyrosDigital, 2018-08-29:
@taltultc I tried your mov - mp4 model but it isn't working on your master, 
looks like the recent commit on the .m file has removed the code. 

Seems there is no current working mov - mp4 solution available that plays well with react-native-camera. 

Thinking of rolling our own, has anybody found a solution for this?