react-native-camera/react-native-camera #301: Invalid number value (NaN) in JSON write (iOS 9.3.2)

CharlieHawker, 2016-05-28:
### Steps to reproduce
1. Create a simple Camera view with onZoomChanged so that it is zoomable (
2. Pinch to zoom repeatedly on native device running iOS 9.3.2 (other versions untested)
3. After a while, app crashes with error:

```
2016-05-28 17:08:48.109 [error][tid:com.facebook.react.JavaScript][RCTJSCExecutor.m:441] Cannot convert argument to string: Error Domain=RCTErrorDomain Code=0 "Invalid number value (NaN) in JSON write" UserInfo={NSLocalizedDescription=Invalid number value (NaN) in JSON write}
2016-05-28 17:08:48.114 [fatal][tid:com.facebook.react.JavaScript] Invalid number value (NaN) in JSON write
```

### Expected behaviour
App should not crash, onZoomChanged should not trigger when there is an error in velocity or zoom calculation.

### Actual behaviour
It seems the velocity and/or zoom calculations happening in **RCTCameraManager.m** can sometimes fail and return NaN values. These are not filtered out causing the app to crash. I added the following line to log the state on line 942 of **RCTCameraManager.m**:

```
NSLog(@"info: %@", event);
```

and the first failure was caused by NaN on the velocity being passed to the zoom function:

```
2016-05-28 17:08:48.107 TinyMagnifier[1614:805674] info: {
    target = 7;
    velocity = nan;
    zoomFactor = nan;
}
```

Thereafter the velocity sometimes changed to another value other than NaN but the zoom always remained NaN in the log entries. The app then crashed.

### Environment
- Node.js version 6.1.0
- React Native version 0.25.1
- iOS 9.3.2

### react-native-camera
**Version**: 0.3.8

CharlieHawker, 2016-05-28:
It seems that AVFoundation can return NaN for velocity and it's something to be weary of. This [stackoverflow comment](http://stackoverflow.com/questions/10220958/avcapturedevice-camera-zoom#comment-59664446) makes reference to it and suggests backing out of any calculations at that point and returning. 

Having said that, I can't find anything on developer.apple.com relating to velocity coming out as NaN.

CharlieHawker, 2016-05-28:
Adding:
```
    if ( isnan(velocity) ) {
      return;
    }
```

to the top of the aforementioned function in **RCTCameraManager.m** seems to resolve the issue for me.

Will create PR.

rt2zz, 2016-05-28:
Awesome, thanks for digging in. Looking forward to the pr!

CharlieHawker, 2016-05-28:
No worries @rt2zz - pr created

commit 61617e2b (2016-05-28): Fixes #301 (#302)
closed 2016-05-28 by rt2zz