react-native-camera/react-native-camera #303: undefined is not an object (evaluating 'PropTypes.oneOfType')

israelidanny, 2016-05-29:
### Steps to reproduce
1. Installed the module according to README
2. Put it into a working simple app I have in a component like so:
```
import React from 'react'
import { StyleSheet, View } from 'react-native'
import paddingTop from '../constants/paddingTop'
import Camera from 'react-native-camera'
const { Component, PropTypes } = React

export default class Capture extends Component {
  static propTypes = {
    navigator: PropTypes.object.isRequired
  };

  render() {
    return (
      <View style={[styles.container, { paddingTop }]}>
        <Camera
          ref="camera"
          style={styles.camera}
        />
      </View>
    )
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'column'
  },
  camera: {
    flex: 1
  }
})
```

### Expected behaviour
The camera module should work

### Actual behaviour
I get the following errors:
```
2016-05-29 19:19:41.273 [error][tid:com.facebook.react.JavaScript] undefined is not an object (evaluating 'PropTypes.oneOfType')
2016-05-29 19:19:41.274 [fatal][tid:com.facebook.react.RCTExceptionsManagerQueue] Unhandled JS Exception: undefined is not an object (evaluating 'PropTypes.oneOfType')
2016-05-29 19:19:41.294 [error][tid:com.facebook.react.JavaScript] Module AppRegistry is not a registered callable module.
2016-05-29 19:19:41.294 [fatal][tid:com.facebook.react.RCTExceptionsManagerQueue] Unhandled JS Exception: Module AppRegistry is not a registered callable module.
```

I don't even have to use the `<Camera />` component to get this, all I need to do is import it. 

### Environment
- **Node.js version**: node v5.6.0
- **React Native version**: react-native@0.26.0
- **React Native platform + platform version**: iOS 

### react-native-camera
**Version**: react-native-camera@0.3.8

israelidanny, 2016-05-29:
I think I have a clue here:
http://stackoverflow.com/questions/37320748/upgrading-to-26-broke-the-dependencies

Apparently starting from `react-native@0.26.0` it should be `var React = require('react')` and not `var React = require('react-native')`

rt2zz, 2016-05-29:
master is pretty unstable. We will be releasing v1 beta soon, but we need a help to get things stable and production ready. 

Not sure what your timeline is but I would install from master for now. We do not have any more breaking changes planned so the api should be pretty stable.

pixelgamma, 2016-06-03:
It seems if you've installed the module from **npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save** you should have a working version. If you've installed from **npm install react-native-camera --save**, you'll get the error you mentioned with RN v0.26.

israelidanny, 2016-06-04:
@rt2zz what *is* the timeline that this component looks like for now?

Is iOS any more stable than Android? 

What kind of help do you need?

charpeni, 2016-06-13:
Can you test with this branch #304, it seems stable on both platforms to me.

israelidanny, 2016-06-14:
Actually, on iOS so far I had absolutely no issues. I'll test #304 later on, but so far it's been fantastic.

NLebedev, 2016-06-16:
I had the same error, running on react-native 0.27 but as @pixelgamma mentioned installing with npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save fixed the issue

logicminds, 2017-07-30:
also receiving this error with react-native 0.46.

```
"dependencies": {
    "react": "16.0.0-alpha.12",
    "react-native": "0.46.4",
    "react-native-camera": "^0.9.4",
    "react-navigation": "^1.0.0-beta.12"
  },
  "devDependencies": {
    "babel-jest": "20.0.3",
    "babel-preset-react-native": "2.1.0",
    "jest": "20.0.4",
    "react-test-renderer": "16.0.0-alpha.12",
    "reactotron-react-native": "^1.12.1"
  },
```

logicminds, 2017-07-30:
https://facebook.github.io/react/docs/typechecking-with-proptypes.html

Should we be importing like `import PropTypes from 'prop-types';`

wootwoot1234, 2017-08-02:
I think #811 PR fixes this.  I'm testing it now....

devSeb, 2017-11-25:
I have this error 
Install prop-types no resolve this.

my package.json
```
"dependencies": {
		"prop-types": "^15.6.0",
		"react": "16.0.0",
		"react-native": "0.50.4",
		"react-native-camera": "^0.6.0"
	},
	"devDependencies": {
		"babel-jest": "21.2.0",
		"babel-preset-react-native": "4.0.0",
		"jest": "21.2.1",
		"react-test-renderer": "16.0.0"
	},
```
error : 
<unknown> react-native-camera/index.js:75:22

miguel-duarte, 2017-11-29:
this worked for me:
in node_modules\react-native-camera\index.js

- removed PropTypes from `import React, { Component, PropTypes } from 'react';`
- add `import PropTypes from 'prop-types';`

in package.json

- add in "dependencies" `"prop-types": "^15.6.0"`

in QRCode.js changed to :

[QRCode.js.txt](https://github.com/lwansbrough/react-native-camera/files/1513596/QRCode.js.txt)

ivansnek, 2018-01-11:
thanks @NLebedev current working with `react-native: ^0.50.0` and using:

`npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save `

fixed my issue too. Also I have to mention that I installed manually not using the `link` command.

sibelius, 2018-01-11:
Fixed on master

closed 2018-01-11 by sibelius

hazeeb, 2018-01-13:
```
react-native: "0.50.3"
react-native-camera: "0.6.0",
```

I'm having the same issue, Installed module with below command.
`npm install react-native-camera --save`
 
Tried importing proptypes, but no use.

nicemanis, 2018-01-15:
This issue is not resolved yet. If you install 0.6.0 version, then you get the prop type error, if you install the git version, then you get the previously mentioned error for cameraview

sibelius, 2018-01-15:
Check RNCameraExample

You need jitpack

Kennytian, 2018-01-18:
@hazeeb
allprojects {
    repositories {
        jcenter()
    +     maven {
    +        url "https://maven.google.com"
    +    }
    }
}

yoesak, 2018-01-24:
I follow miguel-duarte, and it does work. Don't forget to look at attached file QRCode.txt.

douggard, 2018-02-02:
Same error as @hazeeb.