react-native-camera/react-native-camera #306: Collaborators

charpeni, 2016-06-01:
I noticed the lack of collaborators on this project, many pull requests that are really profitable to the package and issues are still open.

I know you have other things to do, that's why I could help you if you want. I'm currently working with this package and I have time to handle some issues and PRs. It's up to you!

cc @lwansbrough @rt2zz

wangzhanji, 2016-06-01:
@charpeni can you write a demo for how to scan use this  project.Waiting for your good news, thank you!

charpeni, 2016-06-01:
@wangzhanji Sure, I can take a look in the next days. 

Until then, you can check this example I made #304.

In a general way, the documentation needs a good help. That's why i'm asking for push access/collaborator, if they want of course.

rt2zz, 2016-06-01:
@charpeni thanks for the contributions! We could definitely use some more manpower maintaining this project. Can you shoot me an [email](https://github.com/rt2zz) so we can figure out the logistics?

charpeni, 2016-06-01:
@rt2zz Done.

E-P1983, 2016-06-02:
@charpeni can you perhaps look into the issue with the RectOfInterest mentioned in https://github.com/lwansbrough/react-native-camera/issues/190 ? I'm really stuck on this :( seems the orientation in iOS devices is wrong...

wangzhanji, 2016-06-03:
@charpeni I found a piece of code, and is already in use.

import React from 'react';

import {
  StyleSheet,
  View,
  Text,
  Dimensions,
  TouchableOpacity,
  Vibration
} from 'react-native'

import Camera from 'react-native-camera';

var QRCodeScreen = React.createClass({

  propTypes: {
    cancelButtonVisible: React.PropTypes.bool,
    cancelButtonTitle: React.PropTypes.string,
    // onSucess: React.PropTypes.func,
    onCancel: React.PropTypes.func,
  },

  getDefaultProps: function() {
    return {
      cancelButtonVisible: false,
      cancelButtonTitle: 'Cancel',
    };
  },

  _onPressCancel: function() {
    var $this = this;
    requestAnimationFrame(function() {
      $this.props.navigator.pop();
      if ($this.props.onCancel) {
        $this.props.onCancel();
      }
    });
  },

  _onBarCodeRead: function(result) {
    var $this = this;

    if (this.barCodeFlag) {
      this.barCodeFlag = false;

      setTimeout(function() {
        Vibration.vibrate();

        alert(result.data);
        // $this.props.onSucess(result.data);
      }, 1000);
    }
  },

  render: function() {
    var cancelButton = null;
    this.barCodeFlag = true;
    
    if (this.props.cancelButtonVisible) {
      cancelButton = <CancelButton onPress={this._onPressCancel} title={this.props.cancelButtonTitle} />;
    }

    return (
      <Camera
        onBarCodeRead={this._onBarCodeRead}
        style={styles.camera}>
        <View style={styles.rectangleContainer}>
          <View style={styles.rectangle}/>
        </View>
        {cancelButton}
      </Camera>
    );
  },
});

var CancelButton = React.createClass({
  render: function() {
    return (
      <View style={styles.cancelButton}>
        <TouchableOpacity onPress={this.props.onPress}>
          <Text style={styles.cancelButtonText}>{this.props.title}</Text>
        </TouchableOpacity>
      </View>
    );
  }
});

var styles = StyleSheet.create({

  camera: {
    height: Dimensions.get('window').height,
    alignItems: 'center',
  },

  rectangleContainer: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
    backgroundColor: 'transparent',
  },

  rectangle: {
    height: 250,
    width: 250,
    borderWidth: 2,
    borderColor: '#00FF00',
    backgroundColor: 'transparent',
  },

  cancelButton: {
    flexDirection: 'row',
    justifyContent: 'center',
    backgroundColor: 'white',
    borderRadius: 3,
    padding: 15,
    width: 100,
    bottom: 10,
  },
  cancelButtonText: {
    fontSize: 17,
    fontWeight: '500',
    color: '#0097CE',
  },
});

module.exports = QRCodeScreen;

closed 2016-06-13 by charpeni