react-native-camera/react-native-camera #309: Android photo taken in portrait gets rotated when saved to camera roll

daavidjobe, 2016-06-02:
### Steps to reproduce
1.Capture an image on a Samsung device in portrait mode and save it to camera roll

### Expected behaviour
The image should be saved as portrait

### Actual behaviour
The image is saved as landscape

This could be fixed by rotating the image if device orientation is 0 and device model is samsung but it´s a bad solution. I don´t know if the issue exists on all samsung devices or just the ones I have tested it on. Also, rotating a bitmap makes the module slow

### Environment
- **Node.js version**:
- **React Native version**: 0.26.2
- **React Native platform + platform version**: Android 5.1

### react-native-camera
**Version**: master

abalhier, 2016-06-02:
Same issue here on a Samasung Galaxy S5. Same RN version.
I don't have another device to try this on though :/

daavidjobe, 2016-06-02:
Forgot to mention what models I´ve tested it on. I´m experiencing this on Galaxy S4 and S6. I think this ultimately comes down to a hardware issue. I don´t think rotating the images on these devices is the right solution since it is very slow and the maintainers would not accept a PR of that kind.

nicastelo, 2016-06-02:
Same issue on LG H340 Leon

daavidjobe, 2016-06-02:
I found a solution that suit my needs at least. I changed the captureTarget to temp and then saving it in my own Java code. This way i can show the temporary image while I´m rotating and what not. The bad part is that I have to maintain some sort of blacklist of models this problem effects.

abalhier, 2016-06-06:
What I ended up doing was to retrieve camera orientation and expose it via a React module like this :
```
    static {
        for (int i = 0; i < Camera.getNumberOfCameras(); i++) {
            Camera.CameraInfo info = new Camera.CameraInfo();
            Camera.getCameraInfo(i, info);
            if (info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT) {
                FRONT_CAMERA_ORIENTATION = info.orientation;
            } else if (info.facing == Camera.CameraInfo.CAMERA_FACING_BACK) {
                BACK_CAMERA_ORIENTATION = info.orientation;
            }
        }
    }

    @Nullable
    @Override
    public Map<String, Object> getConstants() {
        return ImmutableMap.<String, Object>of(
                "BACK_CAMERA_ORIENTATION", BACK_CAMERA_ORIENTATION,
                "FRONT_CAMERA_ORIENTATION", FRONT_CAMERA_ORIENTATION
        );
    }
```

Then in my JS code I could rotate the image :
```
render() {
        const orientation = MyNativeModule.BACK_CAMERA_ORIENTATION;
        return (
            <View style={container}>
                <Image style={{
                    transform: [
                        width: screenWidth,
                        height: screenHeight,
                        {scale: screenHeight/screenWidth,}, //for 90, needs to depend on orientation
                        {rotate: orientation,},
                    ],
                }} source={{uri : this.props.image.path,}} resizeMode='contain' />
            </View>
        );
    }
```

dcworldwide, 2016-06-09:
Same issue on s5

muratozgul, 2016-06-21:
Same issue on OnePlus X

muratozgul, 2016-06-21:
I have this issue on iPhone 4 (8.4.1). Saving temp

cornedor, 2016-06-30:
This probably will be fixed when the Camera2 API isused (http://www.theverge.com/2015/11/9/9696774/google-nexus-5x-upside-down-camera). This however is quite a lot of work since both API's need to be implemented to keep this module compatible with older devices, and this will still not solve all problems. Some devices won't work well with the new API as seen in the release notes of this repo: https://github.com/commonsguy/cwac-cam2.

dcworldwide, 2016-08-04:
https://github.com/lwansbrough/react-native-camera/issues/336

tobiasjunsten, 2016-09-09:
Is this really a bug? The orientation is set correct in the exif orientation tag. If you need to save the image with a neutral rotation a generic solution would be to check the exif rotation tag, rotate the image according to the tag and then set the tag to 1 (normal).

RichardBoyewa, 2016-09-28:
@tobiasjunsten It is. have you worked with this lib before? If you have corrected it kindly share. 👍

alwx, 2016-11-29:
I've fixed this problem (thanks @tobiasjunsten for the idea).
You can check my PR here: https://github.com/lwansbrough/react-native-camera/pull/493

npomfret, 2017-04-26:
@daavidjobe  are you able to test my PR?  It may help with this https://github.com/lwansbrough/react-native-camera/pull/682.

guerrysemones, 2017-06-20:
I see this is not closed despite the PR's listed above. I am running 0.9.4. Our app sets portrait orientation, but  Samsung devices (galaxy S4, S5 and S7) all capture in landscape. I the deprecated fixOrientation to true, and that fixed the problem. Is there a build in the pipeline that fixes this without using fixOrientation? Thanks!

adjustit, 2017-06-26:
I've got the same issue with `0.9.0` (the release went to production about 10 minutes before `0.9.4` was released, d'oh) on several Samsung devices, most notably the Galaxy Tab A.

Chrome on my laptop displays the images as expected once uploaded to my application server, but passing them through any kind of processing library (`GD2`, et al) gives the aforementioned rotation result.

cornedor, 2017-06-29:
@adjustit That is probably something else. A photo's EXIF data contains the orientation, and opening the photo in most photo viewers it will automatically rotate the image. See for example: https://stackoverflow.com/questions/7489742/php-read-exif-data-and-adjust-orientation

adjustit, 2017-07-06:
@cornedor we have narrowed it down to an issue that only seems extant in Samsung devices, apparently it's a relatively common issue on certain classes of Galaxy device. The issue manifests itself by always setting orientation to 0 in the EXIF metadata, so as far as the code is concerned it's interpreting the directionality correctly.

basvdijk, 2017-07-14:
Looks very related to this issue I reported by the cordova-camera-plugin developers: https://issues.apache.org/jira/browse/CB-12958

MelchiorGaspar, 2017-07-16:
I can CONFIRM this issue exists.. when I use the default built in camera app 
that comes with my Samsung Galaxy S7..
only one pic I took, that being the first DID NOT  contain any EXIF data OR the orientation FLAG..
I am pretty sure I used I guess there is a camera app built into Facebook's MESSENGER APP lol...

the other two photos I took used.. the default camera.. X(
UUGHH SAMSUNG!!!!!!!!!!!! 
IDIOT DEVELOPERS!!!!!!!!!!!!!!!

DaniAkash, 2017-08-22:
Is there any way (either fix/hack) to get the orientation fixed in Samsung devices? I'm facing this issue for a long time.

MelchiorGaspar, 2017-08-22:
@DaniAkash  
I copy the images to a PC folder on Windows.. 
then right-click and choose "Rotate Clockwise"

DaniAkash, 2017-08-22:
@MelchiorGaspar 
Is there a way to do it in the Native Java side?

MelchiorGaspar, 2017-08-22:
not that I am aware of.. my method is just a work around...
I can't find any functionality in my Galaxy S7 that rotates the images in a folder...

basvdijk, 2017-08-22:
Another workaround I discovered was letting my customers make a photo with their native cam app and then in my app pick this image from the gallery. It looks like native app sets the correct EXIF orientation.

A think that could speedup things is to vote for the issue I've created for the Cordova plugin at https://issues.apache.org/jira/browse/CB-12958

caiooaugusto, 2017-11-01:
A hot fix is 
allowEdit: true,
correctOrientation: true,

But not working withouy allowEdit: true

:(

harsukhbir, 2017-11-27:
Capture an image on an iPhone device in portrait mode and save it to camera roll Landscape wrong orientation. Did anyone solve this issue?

basvdijk, 2017-11-27:
We fixt it to offer the user to rotate the image after sending it to the server. I don't know if the bug is fixed in version 3.0.0 of the camera plugin

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

seftrick, 2018-08-25:
I'm still have an issue with that!

Bonten156, 2019-06-21:
I have the same kind of issue with face detection with camera2, normally face detection only work in portrait and with a samsung device, it only work in landscape...

Michaelvilleneuve, 2019-07-03:
Hi, still seeing this issue on a Samsung Galaxy Note 3, Android 5.0

Michaelvilleneuve, 2019-07-03:
closing in favor of https://github.com/react-native-community/react-native-camera/issues/1807

closed 2019-07-03 by Michaelvilleneuve