react-native-camera/react-native-camera #310: Photo disappears from app after updating on iOS

Roconda, 2016-06-03:
Hi,

I'm using react-native v0.25 and deploy my app using testflight at the moment. Currently when I store an image using the capture target 'Camera.constants.CaptureTarget.disk', and it all works well. After a new build the image has been removed.

Is this expected behaviour, and what is the best way to solve this issue?

Many thanks

AndreasKlein, 2017-12-16:
We are seeing the same issue. Are there any new on this? @charpeni

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco