react-native-camera/react-native-camera #314: Camera flashMode not working on iphone 6 v9.3.2

DiegoGallegos4, 2016-06-07:
Reproduce: (use demo on ReadMe)
1.Load Camera
2.set FlashMode={Camera.constants.FlashMode.on}

The camera does not trigger flash. My react-native-camera version is "react-native-camera@0.3.8".

```
                         <Camera
		          ref={(cam) => {
		            this.camera = cam;
		          }}
		          style={styles.preview}
		          aspect={Camera.constants.Aspect.fill}
		          flashMode={Camera.constants.FlashMode.on}>
			         <Text>[CAPTURE]</Text>
		        </Camera>

```
- **Node.js version**: 3.73
- **React Native version**: 0.21
- **React Native platform + platform version**: iOS 9.0

phpscrpt, 2016-06-28:
+1 here

charpeni, 2016-09-23:
Should be fixed on master, if not please reopen this issue.

closed 2016-09-23 by charpeni