react-native-camera/react-native-camera #315: Invalid Props error on demo

DiegoGallegos4, 2016-06-07:
This are the following errors when running the demo on the notes: 

1. Warning: Failed propType: Invalid props.style key `color` supplied to `View`.
2. Warning: Failed propType: Invalid props.style key `color` supplied to `RCTView`.

This happens when running the demo on:
1. npm: v3^
2. react-native: v0.21
4. ios 9.3.2

charpeni, 2016-06-13:
Where did you find a demo?

DiegoGallegos4, 2016-06-14:
README's code.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco