react-native-camera/react-native-camera #316: Eslint & CI

charpeni, 2016-06-08:
Hi there,

I'm just wondering, should we add a linter in this repo ? For the sake of consistency between multiple contributors.

I suggest the [Airbnb React/JSX Style Guide](https://github.com/airbnb/javascript/tree/master/react) and it can be setup with [eslint-config-airbnb](https://github.com/airbnb/javascript/tree/master/packages/eslint-config-airbnb), I can do that if you want to.

Furthermore, I think this repo should use a CI like [CircleCI](https://circleci.com/) it could automatically run eslinter and unit tests on each pull requests.

In addition, to improve the pull requests reviewing process, i'm thinking about using the [mention-bot](https://github.com/facebook/mention-bot).

rt2zz, 2016-06-13:
👍 

I am not sure about the specific eslint style we want. Ideally we would pick a config that roughly matches the current code style, or at least requires minimal changes.

lwansbrough, 2016-06-28:
Either Airbnb's React standard or standardjs.com

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco