react-native-camera/react-native-camera #320: open the camera first it is ok,reopen black screen

py110, 2016-06-15:
### Steps to reproduce
1、open the camera from the navigator
2.  first time the camera  show ok but when i back from the navigator then reopen ，the camera is black

`

'use strict';
var React = require('react');
var ReactNative = require('react-native');
var {
  StyleSheet,
  View,
  Text,
  Image,
  Dimensions,
  Platform,
} = ReactNative;

import Camera from "react-native-camera";
var WebPage= require('../yingyong/webPage');

var Qrcode = React.createClass({
  statics: {
    title: '扫码',
    description: '具体的描述',
    id: 'Qrcode',
  },
  componentDidMount() {
    Camera.checkDeviceAuthorizationStatus()
    .then((data)=>{
      console.log("data:"+data);
    })
    .catch((error)=>console.log(error))
  },
  getInitialState() {
    return {
      sectionID:'',
      userString:'',

    };
  },
  openPage(code){
    if (DEV) {
      console.log(code);
    }
    this.camera.stopCapture();
        this.props.navigator.replace({
          title:'详情',
          component:WebPage,
          passProps:{url:url},
          leftButtonIcon: require('../../image/leftarraw@2x.png'),
          onLeftButtonPress: () => this.props.navigator.pop(),
        });
  },
  render() {
    let scanArea = null
    if (Platform.OS === 'ios') {
      scanArea = (
        <View style={styles.rectangleContainer}>
          <View style={styles.rectangle} />
        </View>
      )
    }
    return(
      <Camera
        ref={(cam) => {
           this.camera = cam;
         }}
        onBarCodeRead={(code) =>this.openPage(code)}
        captureMode={Camera.constants.CaptureMode.video}
        captureAudio={false}
        defaultOnFocusComponent={true}
        keepAwake={false}
        style={styles.camera}>
        {scanArea}
      </Camera>
    )
  }
});

const styles = StyleSheet.create({
  camera: {
    flex: 1
  },
  rectangleContainer: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
    backgroundColor: 'transparent'
  },
  rectangle: {
    height: 250,
    width: 250,
    borderWidth: 2,
    borderColor: '#00FF00',
    backgroundColor: 'transparent'
  }
})

module.exports = Qrcode;

`

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**: 5.9.0
- **React Native version**: 0.27.2
- **React Native platform + platform version**: iOS 9.3

JBerendes, 2016-08-01:
Wrap your navigation in a requestAnimationFrame() And make sure you are disabling the camera before you navigate and enabling when you return.

ericchen0121, 2016-08-02:
@JBerendes I'm having this same issue. Can you explain in code how to wrap the navigation in requestAnimationFrame() and also how to disable the camera before you navigate and re-enable? 

I don't see this in the docs at all, so wondering how it's done.

ramsestom, 2016-08-10:
Having the same problem. Anyone can help on how to disable the camera before navigate?

esgudnason, 2016-10-26:
+ 1

jlyons, 2017-01-31:
Any solutions for this?

knapeto, 2017-02-24:
I solved it! You have to use component state. For first open camera it ok BUT before redirecting you need to hide a camera using state.

JBerendes, 2017-02-24:
`
// Set the cameraEnabled state

componentDidMount() {
  requestAnimationFrame(() => {
    // update the camera state here or send a value to a function that changes the cameraEnabled state
  })
}
`

nouh, 2017-05-05:
I still have this problem on android. help!!!

logileifs, 2017-05-18:
I almost managed to solve this problem by wrapping the camera in a component which receives props to either return the camera preview or an empty view. This works on regular navigation since I can pass params to the screen which renders the cam preview to enable the camera but fails with `react-navigation` when going back since the screen doesn't get re-rendered and there is no hook provided to tell the component it has become active again.

Camera wrapper:
```
export default class CameraView extends Component
{
	constructor(props)
	{
		super(props)
	}

	capture(options)
	{
		return this.camera.capture(options)
	}

	render()
	{
		if (!this.props.enabled)
		{
			return (
				<View style={{flex:1, flexDirection: 'column'}}/>
			)
		}

		return (
			<Camera
				ref={(cam) => {
					this.camera = cam;
				}}
				style={{
					flex: 1,
					flexDirection: 'column',
				}}
				captureQuality={Camera.constants.CaptureQuality.medium}
				captureTarget={Camera.constants.CaptureTarget.disk}
				orientation={Camera.constants.Orientation.auto}
				aspect={Camera.constants.Aspect.fill}>
				{this.props.children}
			</Camera>
		)
	}
}
```

@nouh if you still haven't found a work-around the code sample above might help you

logileifs, 2017-05-18:
seems like this problem has been fixed on master, found [this commit](https://github.com/lwansbrough/react-native-camera/commit/f696610115b3ff725de60ed9a0ae6015541f74d3) when looking at the code and after I re-installed from master, I don't need the camera wrapper anymore

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco