react-native-camera/react-native-camera #321: Images are not accessible upon rebuilding ios project

israelidanny, 2016-06-16:
This could be related to:
https://github.com/lwansbrough/react-native-camera/issues/310

When I'm working on a project and save using `disk` mode, everything works fine until I rebuild the project and run it. The files are still there (if I check the actual drive folder), but they "disappear" for the app itself. 

Replicating is easy - just snap some pics via `disk` mode. Save the URI somewhere, and then try to display that URI after you rebuilt the ios project.

ericchen0121, 2016-07-15:
Same issue. Any movement on this?

philipheinser, 2017-02-13:
Same problem here. We also loose all the images after an app update is production...

philipheinser, 2017-02-13:
@israelidanny @ericchen0121 @charpeni This is because iOS App are installed in a Sandbox and after each install the app is in a different directory. So we need to export the path in the document directory and not the absolute url.

AndreasKlein, 2017-12-16:
This is a real issue for us. Are there plans to address this? @philipheinser @Iwansbrough @rt2zz

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco