react-native-camera/react-native-camera #323: reopen the camera screen has problem.

jiachaosun, 2016-06-17:
PLEASE see the mp4 file in zip file.
[react-native-camera-bug.mp4.zip](https://github.com/lwansbrough/react-native-camera/files/319729/react-native-camera-bug.mp4.zip)

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**: 6.2.1
- **React Native version**: 0.27
- **React Native platform + platform version**: iOS 9.3.1

### react-native-camera
**Version**:  "master"

charpeni, 2016-06-17:
Can you provide more information about the issue ?

You should use a software to do a gif instead of a mp4, I don't feel safe with that.

ramsestom, 2016-08-10:
Probably having the same issue. If I take a picture with a camera view, then navigate to another view and go back to the camera view to take a new picture, it appears white (and app is frozen) but I do not see any error in the dev console so I don't know what is the issue...

kesha-antonov, 2016-09-19:
Same.
When I go to screen with camera for first - all ok.
When second, third... - only white screen where should be camera.

kesha-antonov, 2016-09-19:
When I call `@camera.capture()` - it still works and returns `data.path`. (I did not test on real device).

kesha-antonov, 2016-09-19:
Do you have any ideas guys?

marxsk, 2016-09-21:
No idea yet but I have very same problem. This looks like https://github.com/lwansbrough/react-native-camera/issues/111 (from Oct/2015) and https://github.com/lwansbrough/react-native-camera/issues/215 (Feb/2016) where Experimental Navigator is mentioned. You should fine a way how to prevent multiple renders of camera. I'm not sure if javascript will be enough for that :/

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

Bartlebyy, 2018-10-08:
I had the same problem. The camera component would not unmount when I switched between tabs. So I took advantage of the `componentDidMount` and added two listeners `willFocus` and `willBlur` and now it works. 
``` 
 componentDidMount() {
    const { navigation } = this.props;
    navigation.addListener('willFocus', () =>
      this.setState({ focusedScreen: true })
    );
    navigation.addListener('willBlur', () =>
      this.setState({ focusedScreen: false })
    );
  }

  render() {
    const { hasCameraPermission, focusedScreen } = this.state;
    if (hasCameraPermission === null) {
      return <View />;
    } else if (hasCameraPermission === false) {
      return <Text>No access to camera</Text>;
    } else if (focusedScreen){
      return (this.cameraView());
    } else {
      return <View />;
    }
  }
```