react-native-camera/react-native-camera #324: Could not find method... errors in Android

dudeinthemirror, 2016-06-17:
### Steps to reproduce
1. render a Camera component

```
        <Camera
          style={styles.container}
          ref="camera"
          type={this.state.cameraType}
          captureAudio={false}
        >
```

see complete code for the CameraContainer.js file here: https://gist.github.com/dudeinthemirror/e1751f0b52241690533e236f75b22a9e
2. run in Genymotion or tethered device
3. 

### Expected behaviour
Capture button (cameraButton) and flip view button (flipButton) should render on top of the camera view

[image]

### Actual behaviour
Capture and flip buttons do not show (The exact same code works fine in iOS - see above s/shot)

[image]

and these errors show up in the logcat:

https://gist.github.com/dudeinthemirror/3cba9433bcc6e274b878d04c5cf4a50a

### Environment
- **Node.js version**:
5.4.0
- **React Native version**:
react-native-cli: 1.0.0
react-native: 0.26.3
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Android 4.4.4

### react-native-camera
**Version**: npm version or "master"
"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",

dudeinthemirror, 2016-06-18:
Fixed it with something that doesn't quite make sense, but it now works on both iOS and Android:
Took the `TouchableHighlight`'s (cameraButton and flipButton) out from under `<Camera>` and put them at the same level with `<Camera>` within a container `<View>`

```
      return (
        <View style={styles.container}>
         
         <Camera style={styles.camera}
            ref="camera"
            type={this.state.cameraType}
            captureAudio={false}
          >
          </Camera>
          
         <TouchableHighlight
             style={styles.cameraButton}
             onPress={this._takePicture}>
             <View style={styles.cameraButtonInner} />
          </TouchableHighlight>

          <TouchableHighlight
             style={styles.flipButton}
             onPress={this._flipCamera}>
             <Icon allowFontScaling={false}
             name='cycle'
             size={24}
             color={'#FFFFFF'}
          />
          </TouchableHighlight>
        
      </View>
```

error-bruno, 2016-06-21:
Not sure if this will fix your errors in logcat, but there is currently a PR dealing with the Z-index for android. 

[Fix Z-Index](https://github.com/lwansbrough/react-native-camera/pull/308) That fixes [Issue #286](https://github.com/lwansbrough/react-native-camera/issues/286)

dudeinthemirror, 2016-06-21:
thanks @error-bruno, will verify once the PR is in.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco