react-native-camera/react-native-camera #326: App crashes on Code Reload when permission for Microphone is not set

nschoen, 2016-06-20:
### Steps to reproduce
1. Edit permissions: Allow Camera but disallow Microphone usage
2. Run react-native app via Xcode on a physical device (including node server)
3. Open Screen with camera embedded
4. Shuffle iOS device and 'Reload' or if 'Live Reload' is enabled, just save the code
5. App tries to reload JavaScript code

### Expected behaviour
Reload app and continue developing

### Actual behaviour
App crashes, needs to be stopped and rerun via Xcode.
If microphone permission is given, everything is fine. But as I don't need the microphone and for usability reasons that is not really nice. (BTW: is it possible to 'disable' microphone usage?)

[image]

Console says:
```
2016-06-20 17:34:04.154 [info][tid:main][RCTCameraManager.m:244] Error setting property 'captureAudio' of RCTCamera with tag #28: capturing audio
2016-06-20 17:34:04.653 PrototypeApp[1139:575961] Error Domain=AVFoundationErrorDomain Code=-11852 "Cannot use iPad Mikrofon" UserInfo={NSLocalizedDescription=Cannot use iPad Mikrofon, NSLocalizedFailureReason=This app is not authorized to use iPad Mikrofon., AVErrorDeviceKey=<AVCaptureFigAudioDevice: 0x1556fb220 [iPad Mikrofon][com.apple.avfoundation.avcapturedevice.built-in_audio:0]>}
2016-06-20 17:07:30.862 PrototypeApp[1126:569807] *** Terminating app due to uncaught exception 'NSGenericException', reason: '*** AVCaptureSession can't stopRunning between calls to beginConfiguration / commitConfiguration'
*** First throw call stack:
(0x184b80f5c 0x19945bf80 0x1830d15b8 0x1001d5154 0x10098dd70 0x10098dd30 0x10099a7f0 0x100991a20 0x10098dd30 0x10099cf38 0x10099c600 0x199e5d478 0x199e5d028)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) 
```

### Environment
- **Node.js version**: v6.2.1
- **React Native version**: 0.27.2
- **React Native platform + platform version**: iOS 9.0.2 on iPad Mini Retina

### react-native-camera
**Version**: "master

TanTan-TT, 2016-06-29:
+1
Could you give us an interface when permissions of Camera or Micro are disabled? We can alert to users.

JBerendes, 2016-08-01:
Could be just as easy to use https://github.com/yonahforst/react-native-permissions and do the check on app boot or when returning from a backgrounded state. Your app should be forced to boot again if app permissions change. As a side note, with the microphone off the camera goes a bit crazy and changes the proportions of the video in my experience.

wootwoot1234, 2016-08-02:
The first thing I should say is, thanks for this module.  It's worked great for me accept for this bug which is not a show stopper bug but is a bug none the less and should be fixed.

I'm seeing the same thing.  I also noticed that the app crashes when initialized if the ```captureAudio``` prop is not set or is set to true, which I guess is the same thing.

If the microphone is disabled the checkDeviceAuthorizationStatus() returns false.  But I don't use the microphone with the camera because I'm only using it to take pictures.  It should return an object with what is enabled or disabled.

@JBerendes thanks for the link, I think I'll use that.  It's a better solution in my app because of other permissions

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco