react-native-camera/react-native-camera #328: Starting with TorchMode.On is broken

JAStanton, 2016-06-22:
### Steps to reproduce
1. `<Camera torchMode={Camera.Constants.TorchMode.On} />`
2. Observe the flash turn on, then immediately off.

### Expected behaviour
The light should stay on.

### Actual behaviour
The light flickers off.

### Environment
- **Node.js version**: v6.2.1
- **React Native version**: 27.1
- **React Native platform + platform version**: iOS 9.3.2

### react-native-camera
**Version**: npm version or "master": 0.3.8

### Related issue
similar: https://github.com/lwansbrough/react-native-camera/issues/123 and fixed here: https://github.com/lwansbrough/react-native-camera/pull/261 but it doesn't fix the light staying on.  mentioned here as well: https://github.com/lwansbrough/react-native-camera/issues/123#issuecomment-217403124

theopolisme, 2016-07-08:
+1 encountering as well

mesbahsabur, 2017-05-04:
See my comment at #123

wouteralberts, 2017-05-15:
I fixed this by using state for torchMode (initially `off`) and setting it to `on` after showing my camera component:

```
this.setState({showCameraComponent: true}, () => {

    this.setState({torchMode: Camera.constants.TorchMode.on})

});
```
`<Camera ... torchMode={ this.state.torchMode }/>`

Symous, 2017-05-17:
@wouteralberts that's works, thank u!

imapolaris, 2017-09-06:
yes, just as @wouteralberts said. It works well for me!

ma-pe, 2017-11-03:
Thanks for the _workaround_ @wouteralberts. 

Nevertheless this needs to be fixed.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco