react-native-camera/react-native-camera #331: Use Captured Photo on another screen

shonnamartell, 2016-06-27:
I have screen components set up on separate js files. I would like to take the captured image and display it on a new screen. How would I accomplish this with this plugin?

allthetime, 2016-06-27:
When you take a picture the camera returns the image path... Just pass it along to your next view as a prop, or in the navigator push if you are using a navigator... this is what it looks like in my project.
```
	    this.camera.capture()
	      .then((data) => {

	      	var imagePath = data.path;
       
	  	    this.props.navigator.push({
		        name: 'ImageEditor',
		        component: ImageEditor,
		        imagePath: imagePath,
		    });

              })
             .catch((err)=> /* */ )
```
if you're not using a navigator, when you initialize your next view you just need to pass `imagePath` into it somehow.

charpeni, 2016-06-27:
Like @allthetime said, with `imagePath` passed as props you will be able to read the image as base64 and display it inside an `Image` component.

shonnamartell, 2016-06-29:
This is my take picture function
`  takePicture() {
    if (this.camera) {
      this.camera.capture()
        .then((data) => {
          var imagePath = data.path;
            this.props.navigator.push({
                title: 'Add Filter',
                screen: "Gymtastik.FilterScreen",
                imageuri: imagePath,
            });
        })
        .catch(err => console.error(err));
    }
  }`

Now on the receive screen, I don't seem to be getting the image, I am using 

`<Image
      source={{
        isStatic: true,
        uri: 'data:image/jpeg;base64,'+this.props.imageuri,
      }}
      style={{height: 100, width:100}}
      />`

I am very new to react - what I am doing wrong??

allthetime, 2016-06-29:
@shonnamartell did you try doing just `uri: this.props.imageuri` ?

shonnamartell, 2016-06-29:
ok the image seem to be passing now, I had to use `passProps: { imageuri: imagePath },` 

So the image URL `assets-library://asset/asset.JPG?id=9296D0B9-C5D8-4555-88C5-EE8026370A0D&ext=JPG` is now being logged in in console..

HOWEVER... Im still having issues.

The screen is blank when I used this:

`<Image
      source={{
        isStatic: true,
        uri: 'data:image/jpeg;base64,'+this.props.imageuri,
      }}
      style={{height: 100, width:100}}
      />`

and I get a red screen error when I use this: 

`<Image
      source={{
        isStatic: true,
        uri: this.props.imageuri,
      }}
      style={{height: 100, width:100}}
      />`

shonnamartell, 2016-06-29:
I figured it out! I needed `captureTarget: Camera.constants.CaptureTarget.disk` it was saving to the cameraRoll by default. Thanks for all the help!

closed 2016-06-29 by shonnamartell

shoshi, 2017-02-15:
If anybody runs into this issue again and wants to use the default camera roll (`captureTarget: Camera.constants.CaptureTarget.cameraRoll`), there is a solution.

The `RTCCameraRoll` library is *not* included by default. You can go through the instructions [on this page](https://facebook.github.io/react-native/releases/next/docs/cameraroll.html) for [linking the library](https://facebook.github.io/react-native/releases/next/docs/linking-libraries-ios.html) from `/node_modules/react-native/Libraries/CameraRoll`, and then including it along with your `Camera` component.

Your resulting `data.path` will be in the `assets-library://asset/asset.JPG?...` format, and you'll now be able to access the path in an image component:
`<Image source={{uri: myImageVarFromDataPath}}>`

virinchimanepalli, 2021-01-11:
in <Image source={{uri:data:imagepath}} **style={{width=50,height:50}**}> you have to to give style width  nd height  for seeing image , without width nd height ..it shouldn't work in react native. i literally wasted a whole without knowing this