react-native-camera/react-native-camera #336: On IOS picture is always Portrait and on Android its always Landscape

Vani-gurnani, 2016-07-05:
We have set the Camera orientation to default and on IOS it always takes Portrait pictures and on Android it always takes Landscape.

We are in production and this is a really bad issue for our users. Any help will be appreciated.

jakegardner, 2016-07-29:
+1

I have tried all orientation and rotation parameters to no effect.

React Native 0.27.2

jacob-israel-turner, 2016-08-04:
+1 on this, big issue.

dcworldwide, 2016-08-04:
Test:

- On android, simulator and galaxy s5
- Portrait mode set on RN camera
- Phone in portrait mode
- Capture photo
- Display photo in RN image. Displays in Landscape.

This would be a huge win to fix :cry:

jasongaare, 2016-09-21:
For now my workaround is to utilize this library to rotate photos:

https://github.com/bamlab/react-native-image-resizer

dcworldwide, 2016-09-24:
Does anyone have exif rotation logic they could kindly share to show a suggested solution?  Exif across devices can be a bit complicated

TrekSoft, 2017-02-24:
Any update on this issue? I'm having it as well

npomfret, 2017-04-26:
@Vani-gurnani @TrekSoft  are you able to test my PR?  It may help with this https://github.com/lwansbrough/react-native-camera/pull/682.

vagnerSl, 2017-08-25:
+ 1

Gigasz, 2017-08-31:
is this ever gonna be fixed?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco