react-native-camera/react-native-camera #337: Cannot display image (Android)

jinqkim1, 2016-07-06:
I've successfully installed the package and followed the installation guide in README carefully. I used the code from Example.js and I got it working right, but the problem is displaying image.. This is my code below.

import React from 'react';
import Icon from 'react-native-vector-icons/Ionicons';
import RNFS from 'react-native-fs';
const Main = require('./Main.js');
import {
  Image,
  StatusBar,
  StyleSheet,
  TouchableOpacity,
  View,
  Text,
} from 'react-native';
import Camera from 'react-native-camera';

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  overlay: {
    position: 'absolute',
    padding: 16,
    right: 0,
    left: 0,
    alignItems: 'center',
  },
  topOverlay: {
    top: 0,
    flex: 1,
    flexDirection: 'row',
    justifyContent: 'space-between',
    alignItems: 'center',
    backgroundColor: 'rgba(0,0,0,0.4)',
  },
  bottomOverlay: {
    bottom: 0,
  },
  captureButton: {
    flex: 1,
    padding: 15,
    backgroundColor: 'white',
    borderRadius: 40,
  },
  typeButton: {
    padding: 5,
  },
  flashButton: {
    padding: 5,
  },
});

class RunCamera extends React.Component {
  constructor(props) {
    super(props);
    this.camera = null;
    this.state = {
      imageData: null,
      camera: {
        aspect: Camera.constants.Aspect.fill,
        captureTarget: Camera.constants.CaptureTarget.disk,
        type: Camera.constants.Type.back,
        orientation: Camera.constants.Orientation.auto,
        flashMode: Camera.constants.FlashMode.auto,
      },
    };
    this.takePicture = this.takePicture.bind(this);
    this.switchType = this.switchType.bind(this);
    this.switchFlash = this.switchFlash.bind(this);
  }

  takePicture() {
    var self = this;
    if (this.camera) {
      this.camera.capture()
        .then((data) => {
            RNFS.readFile(data.path.substring(7), "base64")  //substring(7) -> to remove the file://
            .then(res => this.setState({imageData: res}));
        })
        .catch(err => console.error(err));
    }
  }

  switchType() {
    let newType;
    const { back, front } = Camera.constants.Type;

    if (this.state.camera.type === back) {
      newType = front;
    } else if (this.state.camera.type === front) {
      newType = back;
    }

    this.setState({
      camera: {
        ...this.state.camera,
        type: newType,
      },
    });
  }

  switchFlash() {
    let newFlashMode;
    const { auto, on, off } = Camera.constants.FlashMode;

    if (this.state.camera.flashMode === auto) {
      newFlashMode = on;
    } else if (this.state.camera.flashMode === on) {
      newFlashMode = off;
    } else if (this.state.camera.flashMode === off) {
      newFlashMode = auto;
    }

    this.setState({
      camera: {
        ...this.state.camera,
        flashMode: newFlashMode,
      },
    });
  }

  get flashIcon() {

    const { auto, on, off } = Camera.constants.FlashMode;

    if (this.state.camera.flashMode === auto) {
      return (<Text>Auto</Text>);
    } else if (this.state.camera.flashMode === on) {
      return (<Icon name = "ios-flash" size = {22} color="#fff"/>);
    } else if (this.state.camera.flashMode === off) {
      return (<Icon name = "ios-flash-outline" size = {22} color="#fff"/>);
    }
  }

  render() {
    if(this.state.imageData){
      console.log(this.state.imageData.slice(0,20));
      return (
        <Image
          source={{uri: 'data:image/jpeg;base64,'+this.state.imageData}}
        />
      );
    }
    return (
      <View style={styles.container}>
        <StatusBar
          animated
          hidden
        />
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={this.state.camera.aspect}
          captureTarget={this.state.camera.captureTarget}
          type={this.state.camera.type}
          flashMode={this.state.camera.flashMode}
          defaultTouchToFocus
        />
        <View style={[styles.overlay, styles.topOverlay]}>
          <TouchableOpacity
            style={styles.typeButton}
            onPress={this.props.navigator.pop}
          >
            <Icon name = "md-close-circle" size = {22} color="#fff"/>
          </TouchableOpacity>
          <TouchableOpacity
            style={styles.typeButton}
            onPress={this.switchType}
          >
            <Icon name = "md-sync" size = {22} color="#fff"/>
          </TouchableOpacity>
          <TouchableOpacity
            style={styles.flashButton}
            onPress={this.switchFlash}
          >
            {this.flashIcon}
          </TouchableOpacity>
        </View>
        <View style={[styles.overlay, styles.bottomOverlay]}>
          <TouchableOpacity
            style={styles.captureButton}
            onPress={this.takePicture}
          >
            <Icon name = "md-recording" size = {22} color="#fff"/>
          </TouchableOpacity>
        </View>
      </View>
    );
  }
}

RunCamera.propTypes = {
    loginInfo : React.PropTypes.object,
    username : React.PropTypes.string,
    email : React.PropTypes.string,
    userid : React.PropTypes.string,
    gender : React.PropTypes.string,
};

module.exports = RunCamera;

As you can see above, the last thing I tried was using react-native-fs, and I also tried using the whole data path instead of taking the first 7 letters out. I'm using Camera.constants.CaptureTarget.disk, but I also tried with camera roll and temp. Also, I've looked at similar issues in this github and tried every solution but I still can't get it working.  All I am getting is a blank page. I don't get any errors but I am not getting anything. What am I doing wrong?

p.s. - What I'm trying to do is show the captured picture to the user once they take the picture.

jinqkim1, 2016-07-06:
I also added read external storage permission in my manifest file but still didn't work.

jinqkim1, 2016-07-06:
BTW, capture is returning Object {path: "file:///storage/emulated/0/Pictures/RCTCameraModule/IMG_20160706_185559.jpg"}
just in case people are curious.. I've been trying to display captured image for days now, and I REALLY need some help,, Can ANYONE share some insights on this??

scerelli, 2016-07-08:
@jinqkim1 

```
let data = {path: "file:///storage/emulated/0/Pictures/RCTCameraModule/IMG_20160706_185559.jpg"} // this return from capture promise
<Image source={data.path} />

```

jinqkim1, 2016-07-09:
@scerelli Thank you but that was actually what I first tried to do. I got the path from captured data, but it still did not work.

th3no0b, 2016-08-03:
I'm also having this problem, camera.capture() returns `{"path": "file:///storage/emulated/0/Pictures/RCTCameraModule/IMG_whatever.jpg"}`, but using `<Image source={{uri: data}} />` or `<Image source={{uri: data.path}} />` doesn't work.

th3no0b, 2016-08-03:
Figured out the problem 30 seconds after I posted this. Make sure you have a width and height specified on your image!

`<Image source={{uri: data.path}} style={{width: 500, height: 500}} />`

jinqkim1, 2016-08-03:
@th3no0b  thank you. That was it.. Sorry for the stupid question.

closed 2016-08-03 by jinqkim1

XHTeng, 2016-09-20:
@th3no0b I also has the problem to waste a day，aha...thans!